FreeCAD/FreeCAD #6922: [Feature Request] An option to change FEM Pipeline units

FEA-eng, 2022-05-27:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=18&t=68502

### Subproject(s) affected?

FEM

### Idea description

Since 0.20 release, FEM Pipeline legends are always in SI units so displacements are in m, stresses in Pa and so on. However, in many cases, it is much more clear to operate on SI(mm) units, for example  (displacements in mm, stresses in MPa). Thus, it would be good to add an option to change the units of the displayed FEM result legends.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-05-27:
Yes, now that FEM can do multiphysics this is highly desirable.

luzpaz, 2023-02-18:
Hi @donovaly, what about this ticket, is it still relevant?

donovaly, 2023-02-18:
This is still relevant and on my ToDo list.

bdlabitt, 2023-05-09:
It's very awkward to display stress when a system is pressurized to 30 MPa, which is common for high pressure air, the display is very busy and the font is too small.  I'm trying to see stresses of 450-1000 MPa, viewing 4.500000e+8 Pa or whatever is rather silly.  It makes this module seem a bit amateur in it's capabilities, especially when shown publicly as compared to other FEA displays.  At least that is the feedback I am getting, attempting to show that FreeCAD can indeed do FEA.

I don't know the project.  On the surface, the changes don't sound that hard.  Divide by 1e6 and display.  At least that is what I used to do with MatplotLib when I wanted more professional legends.  Is it far more complicated than that?  (Wouldn't surprise me if it was.)

Once the numbers are on the order of three digits, it would be nice to slightly improve the font for readability.  I am no longer young, my visual acuity is poorer than in my youth.  The current font is hard to read.  

Would be good to push this up a bit.  It is nearly a year old.  Thanks.