FreeCAD/FreeCAD #6928: [Bug] <title>

smurfix, 2022-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=19&t=68988&p=597813#p597813

### Version

0.19 (Release)

### Full version info

```shell
Hi,

so I'm in the Parts module. I add a cube, the position of which is based on some formula or other.

Then I go to Parts Design and click Dolly to create a clone, then move the clone around, based on some other formula.

The problem here is that the position of the clone's original shape is stored in the copy *and* it's based on absolute values, not a formula. Thus when I change some of the variables which my positioning is based on, the original moves where it's supposed to, but the copy doesn't beause its placement is based on the calculated position of the original, not on
```

### Issue description

Hi,

so I'm in the Parts module. I add a cube, the position of which is based on some formula or other.

Then I go to Parts Design and click Dolly to create a clone, then move the clone around, based on some other formula.

The problem here is that the position of the clone's original shape is stored in the copy *and* it's based on absolute values, not a formula. Thus when I change some of the variables which my positioning is based on, the original moves where it's supposed to, but the copy doesn't because its placement is based on the calculated position of the original, not on 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-05-29:
@smurfix Needs to be tested on v0.20dev

smurfix, 2022-05-29:
Tested on current 0.20 AppImage. Still a problem.

luzpaz, 2022-05-29:
Please paste the full [About](https://wiki.freecad.org/About) info of the AppImage you're running. Thanks

luzpaz, 2022-05-29:
This mixes Part and PartDesign workbenches. Do you know how to do that ? There is a difference.

adrianinsaval, 2022-05-29:
This would be a feature request rather than a bug, AFAIK clones don't follow the placement of their base feature. Consider using a subshapebinder instead.

djurodrljaca, 2023-08-12:
@luzpaz I have the same problem and I am doing everything in PartDesign workbench.

In my case I have a clone of a body attached to one end of a pipe which then I rotate and move to the other end of the pipe - lets call that Clone1. Then I clone Clone1 and rotate it along the same edge of the pipe as Clone1.

I noticed the problem when I changed the length of the pipe - Clone1 was moved accordingly, but Clone2 was not moved to the edge of the (now longer) pipe.

As already mentioned by @smurfix the problem is that a clone contains the calculated value of the original body's placement formula instead of copying the formula.

There is of course a simple (but annoying) workaround if you are aware of the problem - just replace the placement values of the "clone of a clone" with the original clone's expression.

djurodrljaca, 2023-08-12:
As far as I see it, this is a valid use case but I am relatively new to FreeCAD and don't know the potential problems this would cause elsewhere.

Since I am an experienced C++/Qt developer I would be willing to implement this feature if it is actually a valid use case.

@luzpaz please let me know if I should implement this and if possible please point me to the code responsible for cloning an object so that I can get started faster.

luzpaz, 2023-08-12:
@djurodrljaca come participate in our hackathon and ask your pertinent questions: https://discord.com/channels/870877411049357352/870877411623985154

adrianinsaval, 2023-08-17:
please attach example files because I'm having trouble understanding the usefulness of having superimposed clones. They are not meant to follow placement changes so I'm leaning towards closing this issue

djurodrljaca, 2023-08-17:
@adrianinsaval I think I was probably doing it wrong anyway and I did find a better way of achieving the same result (i.e., ended up doing features all in the same body instead of having multiple bodies). As far as I am concerned this can be closed, I just volunteered to fix/implement it since it seemed simple to do and potentially useful.

**Just to explain the situation I had:**
I had a pipe that I wanted to place 3 "attachments" (rectangles with holes for bolts and nuts). I created the first attachment as a body and then made 2 copies of it rotated by 120° and 240° on the same edge of the pipe (the first level of clones).

Then I cloned the original and rotated it by 180° (it had a "nut shaped hole" so I had to turn it around) and translated it to the other end of the pipe and then cloned it 2 times and again rotated them on the edge of the pipe.

At a later point I made the pipe longer (using a variable from a spreadsheet) and only the clone of the original was moved to the end of the pipe, but the clones of clones remained at their original location since they cloned their parent's translation to the other end of the pipe with the resulting pipe length at the time of cloning and not the expression which contained the pipe length variable reference.

If you think this might be of any use I can recreate the situation and attach it, but otherwise I have no strong need to keep this open.

luzpaz, 2025-05-17:
Related?  
* #17313