FreeCAD/FreeCAD #6933: [Feature Request] Support of multiple licenses per file

wwmayer, 2022-05-29:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=13&t=69030

### Idea description

Sometimes people want to share a project file not only under a single license but multiple licenses. At the moment this is not possible and there is only the workaround to use the comment field of a document.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2022-05-29:
Suggestion: add a property AdditionalLicenses of type PropertyMap and extend the class DlgProjectInformationImp accordingly.

AdarshRawat1, 2023-03-18:
Can I work on this feature?

wwmayer, 2023-03-18:
Sure.

In order to be backward compatible I recommend to add a new property of type App::PropertyStringList. The name of the property can be AdditionalLicences, FurtherLicenses, MultiLicense, ... or something like that and will be filled if more than one license is used.

The currently used property License must not be changed or removed as otherwise older FreeCAD versions won't be able to show the correct license of files created with a newer version.

Then DlgProjectInformationImp should be extended to conveniently set additional licenses

gauriimaheshwarii, 2023-03-18:
@wwmayer Which file should be modified to add the feature?

wwmayer, 2023-03-18:
src/App/Document.cpp and .h

yorikvanhavre, 2023-03-20:
Can I ask in what kind of scenario would someone want multiple license per file?

hashoak, 2023-06-10:
I'm currently working on the issue. Should I also design the UI part? or just the backend part is sufficient?

luzpaz, 2023-10-01:
ping @hashoak