FreeCAD/FreeCAD #6935: [Bug] navicube location sometimes jumps to other side

donovaly, 2022-05-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=69003

### Version

0.19 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: German/Germany (de_DE)
```

### Issue description

- set the NaviCube location to TopLeft
- grab the right border of the main window and shrink the window it to the minimum
result: The NaviCube jumps to the right edge and stays there:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-01:
Interestingly the bug dos not appear on all systems. For example I took the same installer and on some PCs I see this bug while on others I don't. All PCs have Win 10 2021 H2. So it must be something with the graphics settings.

luzpaz, 2022-06-27:
@donovaly was this bug fixed before release?

donovaly, 2022-06-27:
> @donovaly was this bug fixed before release?

No.

Jolbas, 2023-04-03:
I'm working on https://github.com/FreeCAD/FreeCAD/issues/6818 and this issue is going to be fixed at the same time

PR #9356 (2023-04-24): Refactor Navigation Cube
This PR does not conform to the one issue recommendation. Hopefully you can bear reviewing it anyway.
Fixes #9262 
Fixes #6935 
Fixed #6818 
But also a non registered issue from the forum:
[https://forum.freecad.org/viewtopic.php?t=77768](url)
Will hopefully also fix a problem on Raspberry Pi
[https://forum.freecad.org/viewtopic.php?t=76438](url)

Here is a list of changes
- Uses std commands in menu coming with icons
- Cube only draggable if activated in menu 
- [image]
- Ortographic and Perspective reflected in Navigation Cube
- [image]
- [image]
- Remove textures ex [... cut]
closed 2023-04-24 by chennes