FreeCAD/FreeCAD #6963: [Bug] [TD] Dimension Extensions Translation and Combine Functions

WandererFan, 2022-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://www.forum.freecadweb.org/viewtopic.php?p=599447#p599447

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: d492ac30838e4fb07b1ce4b378730f2fb38b9ba3
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.54b
  * Curves 0.3.0
  * fasteners 0.3.40
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

There are a number of text literals and command names that are difficult for translator due to abbreviations.  
Also the toolbar is very large and there are similar tools that could be combined (ex position vertical and horizontal dimensions).

### Anything else?

Work involved:
- change literals and command names to use full words.
- rename icons to match revised text and upload to wiki
- update wiki with new command and icon names

- combine similar functions 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-04-15:
@benj5378 any interest in tackling this issue?

luzpaz, 2023-11-01:
@kaktusus if we can find translators that have dev skills, this could be a ticket for them

WandererFan, 2024-06-10:
fixed as part of #14216.

closed 2024-06-10 by WandererFan