FreeCAD/FreeCAD #6973: [Bug] [FEM] adding a pipeline filter changes transparency

donovaly, 2022-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=18&t=69209

### Version

0.19 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
```

### Subproject(s) affected?

FEM

### Issue description

- set the transparency of a pipeline object to e.g. 75
- select this pipeline and add a filter of your choice
result: the transparency of the pipeline is lost

This bug only occurs if the pipeline shows no field yet (Field is "None").

This is a bit annoying because the transparency was set that one can see the new filter result immediately.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marioalexis84, 2024-03-14:
@FEA-eng can you reproduce this?

FEA-eng, 2024-03-14:
> @FEA-eng can you reproduce this?

No, I can't. The pipeline keeps its transparency after a new filter is added. Actually, I was planning to test it after your fix for another transparency issue was merged since I thought that maybe it could resolve this issue as well but it seems that it is resolved already. So I think that the issue can be closed if you agree.

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36391 (Git)
Build type: Release
Branch: main
Hash: 52fffaf50ef8facea2ecfde23d6f1a7d3ad3d80c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

marioalexis84, 2024-03-14:
> So I think that the issue can be closed if you agree.

Yes.
@chennes, this issue can be closed.

FEA-eng, 2024-03-17:
@maxwxyz Can you close it ?

closed 2024-03-17 by maxwxyz