FreeCAD/FreeCAD #6985: [Bug] Copy & Paste of Part Design feature pastes feature outside the Body object

macdroid53, 2022-06-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=600227#p600227

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.20.29055 (Git)
Build type: Release
Branch: master
Hash: b9b21ddac562f03cef162bb1f7feb4eea3de8b1e
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, OCC 7.7.0.dev
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.12.5
  * Manipulator 1.4.5
  * DynamicData 2.46.0
  * fasteners 0.3.43
  * CubeMenu
  * fcgear 1.0.0
  * plugins
  * Pyramids-and-Polyhedrons
  * Defeaturing
  * Assembly4 0.11.12
  * OpticsWorkbench 1.0.8
  * workfeature-macro
  * sheetmetal 0.2.49
  * A2plus 0.4.56
  * CurvedShapes 1.0.3
  * Curves 0.4.4
  * workfeature
  * lattice2 1.0.0
[/code]
```

### Issue description

Create a Body, add sketch, Pad.
Select Pad Ctrl-C Ctrl-v
Pad is placed in the tree out side the Body.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-06-07:
This sounds like a dupe of an already opened ticket. But I can'r recall which ATM.

macdroid53, 2022-06-07:
As I noted, searching here didn't turn up anything that sounded similar...but, there are probably a hundred search terms I didn't think of...

donovaly, 2022-06-19:
I cannot find a duplicate of this issue.

wwmayer, 2023-09-24:
Related to https://github.com/FreeCAD/FreeCAD/issues/6741

PaddleStroke, 2024-12-05:
+1, I was using using freecad for a change (to be honest I'm not using it so often these days) and came upon this issue and 6741 as well. Very annoying.

macdroid53, 2024-12-05:
This has been a long time issue with copy/paste.