FreeCAD/FreeCAD #6988: [Bug] [Techdraw] Rich texteditor Techdraw icons not readable on dark themes.

MisterMakerNL, 2022-06-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=599460#p599460

### Version

0.20 (Development)

### Full version info

```shell
Not version depended. 
But tested in :
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.28886 (Git)
Build type: Release
Branch: master
Hash: a564d5ced7aecf320a0ac2d73f824a3dc7861eb3
Python 3.8.10, Qt 5.15.3, Coin 4.0.1, OCC 7.5.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.0.1
  * FeedsAndSpeeds 0.4.0
  * FEM_FrontISTR 0.1.0
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0
  * SelectorToolbar
  * sheetmetal 0.2.49
  * ToolbarStyle
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

The text icons in the Rich text editor are not readable in dark theme's since they are black and have no background.
It seems like the background is not addressable directly by a style-sheet, which would be nice.
But for now I think it is just fast and simplest too just add a white background too the icons.
**And I already did that see below for attachments.**
Icons need to be in : [FreeCAD](https://github.com/FreeCAD/FreeCAD/tree/52c86de3cbb62346c94539696a3288a9c28b5f46)/[src](https://github.com/FreeCAD/FreeCAD/tree/52c86de3cbb62346c94539696a3288a9c28b5f46/src)/[Mod](https://github.com/FreeCAD/FreeCAD/tree/52c86de3cbb62346c94539696a3288a9c28b5f46/src/Mod)/[TechDraw](https://github.com/FreeCAD/FreeCAD/tree/52c86de3cbb62346c94539696a3288a9c28b5f46/src/Mod/TechDraw)/[Gui](https://github.com/FreeCAD/FreeCAD/tree/52c86de3cbb62346c94539696a3288a9c28b5f46/src/Mod/TechDraw/Gui)/[Resources](https://github.com/FreeCAD/FreeCAD/tree/52c86de3cbb62346c94539696a3288a9c28b5f46/src/Mod/TechDraw/Gui/Resources)/[icons](https://github.com/FreeCAD/FreeCAD/tree/52c86de3cbb62346c94539696a3288a9c28b5f46/src/Mod/TechDraw/Gui/Resources/icons)/MRTE/

### Anything else?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[srcModTechDrawGuiResourcesiconsMRTE_whiteback.zip](https://github.com/FreeCAD/FreeCAD/files/8859051/srcModTechDrawGuiResourcesiconsMRTE_whiteback.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2022-06-08:
[image]
Screenshot about the issue

WandererFan, 2022-06-10:
merged to master.  commit 1b65b3f600.

closed 2022-06-10 by WandererFan