FreeCAD/FreeCAD #6992: Supporting Qt6

luzpaz, 2022-06-08:


luzpaz, 2022-06-08:
Transferred to FreeCAD/FreeCAD repo

luzpaz, 2022-06-25:
Added the Also [Porting from PySide2 to PySide6](https://doc.qt.io/qtforpython/porting_from2.html) link to the OP of this ticket.

luzpaz, 2022-07-14:
Started tracking Qt6 commits to master in the OP. See **Commits** section

luzpaz, 2022-07-16:
Hi @wwmayer, how would you feel about linking Qt6 related commits to this ticket so we can track them in one place?

wwmayer, 2022-07-16:
339bb73e9f23f is not directly related to Qt6 but allows to reduce some `if QT_VERSION < QT_VERSION_CHECK(...)` checks in client code.

luzpaz, 2022-11-08:
Related: 
#7712  
#7677  
#7647

iamdbychkov, 2022-11-17:
Is there any instruction on installing FreeCad on newer distributions? Official binaries is not useful as every operation fails on some python error.
Is there a way to instruct compile which python to use when building FreeCad from source? For some reason page with building instructions is unavailable.
Shouldn't this instruction be on a front page for now?

adrianinsaval, 2023-05-30:
what's still missing for qt6 support?

berniev, 2023-05-31:
MacAppBundle/CMakeLists.txt
Resolve Homebrew v Pip v native on macOS esp re Python, PySide, Shiboken

luzpaz, 2023-07-23:
Added https://github.com/FreeCAD/FreeCAD/pull/9977 to the OP

luzpaz, 2023-08-06:
We're gonna need some help porting Snap to Qt6 (https://github.com/FreeCAD/FreeCAD-snap/issues/19). Volunteers for this would be very much appreciated!

chennes, 2023-10-23:
FYI compiling against Qt6 with MSVC activates the `/permissive-` option, which we don't currently support. PR #11014 will address these issues.

luzpaz, 2023-12-24:
> a nightly F(edora)39 build at this point isn't going to happen against the stock F39 repos, as there are no python qt5 bindings. For a FreeCAD nightly to be compiled successfully for F39, python qt6 bindings need to be in the virtual environment (there is already a build in copr for python-pyside6). This would require copr to include python-pyside6 in it's virtual build environment for FreeCAD nightly (it can be done, but I don't have enough experience with copr to set it up properly). When (and if) FreeCAD nightly is compiled successfully, any user who enables the FreeCAD nightly repo and wants to use it, would need to install python-pyside6 too - either will be provided from another copr repo, or ideally shipped in the FreeCAD copr repo (until Fedora officially includes the python-pyside6 in their official repos). Can someone just fix the references in the spec file from master to main, so we can get at least an attempt for a build for F38. It might finish successfully, or it might fail, but at least we can get the logs.

ref: https://github.com/FreeCAD/FreeCAD/issues/11619#issuecomment-1868390068

adrianinsaval, 2023-12-25:
Are they using Py_LIMITED_API in their pyside build? Pyside6 uses it by default I think, I had to disable it to be able to build freecad on flatpak because freecad doesn't support that.

ipatch, 2024-08-02:
can someone explain to me how a workbench such as draft which heavily uses the runtime dep of pyside works with qt6?

on the build box i'm working with i have qt6 and pyside6 installed but i still see references in the freecad source using `from PySide.QtCore` which from what i'm experience that "call" will not work with pyside6 because the module name is PySide6, no?

and doing a search in the draft sub dir of the freecad src repo i see all the mentions of `PySide`

```
╰─λ rg -L -i pyside .                                                                                                   0 (0.015s)
./draftutils/todo.py
39:import PySide.QtCore as QtCore

./draftguitools/gui_arrays.py
30:from PySide.QtCore import QT_TRANSLATE_NOOP

./WorkingPlane.py
39:from PySide.QtCore import QT_TRANSLATE_NOOP
1731:            from PySide import QtGui
1768:    from PySide import QtWidgets

./drafttaskpanels/task_polararray.py
30:import PySide.QtGui as QtGui
31:from PySide.QtCore import QT_TRANSLATE_NOOP
87:        self.form = Gui.PySideUic.loadUi(ui_file)

./DraftGui.py
42:import PySide.QtCore as QtCore
43:import PySide.QtGui as QtGui
44:import PySide.QtWidgets as QtWidgets

./drafttaskpanels/task_scale.py
31:import PySide.QtCore as QtCore
32:import PySide.QtGui as QtGui
33:import PySide.QtWidgets as QtWidgets

./draftguitools/gui_annotationstyleeditor.py
30:import PySide.QtGui as QtGui
31:import PySide.QtWidgets as QtWidgets
33:from PySide.QtCore import QT_TRANSLATE_NOOP
62:    form: PySide.QtWidgets.QDialog
94:        self.form = Gui.PySideUic.loadUi(ui_file)

./draftutils/gui_utils.py
51:    from PySide import QtCore
52:    from PySide import QtGui
53:    # from PySide import QtSvg  # for load_texture
710:        # from PySide import QtGui, QtSvg

./drafttaskpanels/task_shapestring.py
31:import PySide.QtCore as QtCore
32:import PySide.QtGui as QtGui
52:        self.form = Gui.PySideUic.loadUi(":/ui/TaskShapeString.ui")

./draftutils/grid_observer.py
34:    from PySide import QtWidgets

./draftutils/init_tools.py
33:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftguitools/gui_togglemodes.py
36:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftguitools/gui_polygons.py
35:from PySide.QtCore import QT_TRANSLATE_NOOP

./drafttaskpanels/task_orthoarray.py
30:import PySide.QtGui as QtGui
31:from PySide.QtCore import QT_TRANSLATE_NOOP
87:        self.form = Gui.PySideUic.loadUi(ui_file)

./draftutils/units.py
[... 307 lines cut ...]

./draftobjects/polygon.py
31:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftobjects/layer.py
31:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftobjects/bspline.py
30:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftobjects/rectangle.py
30:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftobjects/bezcurve.py
30:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftobjects/text.py
31:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftobjects/point.py
31:from PySide.QtCore import QT_TRANSLATE_NOOP

./draftobjects/pointarray.py
31:from PySide.QtCore import QT_TRANSLATE_NOOP
```

chennes, 2024-08-02:
FreeCAD provides a wrapper called `PySide` which, when imported, correctly imports either PySide6 or PySide2 depending on which version of Qt FreeCAD was compiled against. It also provides some other wrapping functions to help mod developers write code that works with both Qt5 and Qt6.

ipatch, 2024-08-02:
> FreeCAD provides a wrapper called `PySide` which, when imported, correctly imports either PySide6 or PySide2 depending on which version of Qt FreeCAD was compiled against. It also provides some other wrapping functions to help mod developers write code that works with both Qt5 and Qt6.

okay if that's the case then looking at the install structure / layout, should the local `PySide` be accessible via the python interpreter?

```
╰─λ cd MacOS/                                                                                                                                                       0 (0.015s)
╭─chris ssh Chriss-Mac-mini in /Users/chris/.local/fc-git/MacOS
╰─λ l                                                                                                                                                               0 (0.013s)
total 680
drwxrwxr-x   5 chris  staff   160B Aug  2 14:21 ./
drwxrwxr-x   8 chris  staff   256B Aug  2 14:21 ../
-rwxr-xr-x   1 chris  staff   194K Aug  2 14:17 FreeCAD*
-rwxr-xr-x   1 chris  staff   142K Aug  2 14:17 FreeCADCmd*
drwxr-xr-x  11 chris  staff   352B Aug  2 14:21 PySide/
╭─chris ssh Chriss-Mac-mini in /Users/chris/.local/fc-git/MacOS
╰─λ tree PySide/                                                                                                                                                    0 (0.014s)
PySide/
├── QtCore.py
├── QtGui.py
├── QtNetwork.py
├── QtSvg.py
├── QtSvgWidgets.py
├── QtUiTools.py
├── QtWebEngineWidgets.py
├── QtWidgets.py
└── __init__.py

1 directory, 9 files
```

luzpaz, 2024-08-28:
Related:  
* #16047

luzpaz, 2024-10-19:
Should we keep this ticket open ?

chennes, 2024-10-19:
IMO no, it should be migrated to individual issues that need addressing: in bulk, FreeCAD works fine with Qt6 (I use it every day!). The catch-all is no longer necessary.

closed 2024-10-20 by luzpaz