FreeCAD/FreeCAD #7000: [Bug] Freecad 0.20 won't start on windows

DoumeDuSud, 2022-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

None

### Version

0.20 (Development)

### Full version info

```shell
None, Freecad.exe and FreeCADCmd.exe don't start

I t'y this version of Freecad : https://github.com/FreeCAD/FreeCAD/tree/0.20beta1 with this asset : FreeCAD-0.20.0beta-WIN-x64-installer-5.exe

I'm on Windows 10 Entreprise version 1803
```

### Issue description

When I try to start Freecad by execute Freecad.exe nothing happen

When I run `"C:\Program Files\FreeCAD 0.20\bin\FreeCADCmd.exe" -v` I have the following output : 

```
Python path configuration:
  PYTHONHOME = 'C:\Program Files\Python310\python.exe'
  PYTHONPATH = 'C:\Program Files\Python310\Lib\site-packages'
  program name = 'C:\Program Files\FreeCAD 0.20\bin\FreeCADCmd.exe'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = 'C:\\Program Files\\FreeCAD 0.20\\bin\\FreeCADCmd.exe'
  sys.base_prefix = 'C:\\Program Files\\Python310\\python.exe'
  sys.base_exec_prefix = 'C:\\Program Files\\Python310\\python.exe'
  sys.executable = 'C:\\Program Files\\FreeCAD 0.20\\bin\\FreeCADCmd.exe'
  sys.prefix = 'C:\\Program Files\\Python310\\python.exe'
  sys.exec_prefix = 'C:\\Program Files\\Python310\\python.exe'
  sys.path = [
    'C:\\Program Files\\Python310\\Lib\\site-packages',
    'C:\\Program Files\\FreeCAD 0.20\\bin\\python38.zip',
    'C:\\Program Files\\Python310\\python.exe\\DLLs',
    'C:\\Program Files\\Python310\\python.exe\\lib',
    'C:\\Program Files\\FreeCAD 0.20\\bin',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 914, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 1407, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1379, in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1510, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1553, in _fill_cache
OSError: [WinError 87] ParamÃ¨tre incorrect: 'C:\\Program Files\\FreeCAD 0.20\\bin'
```

It's not working also without  PYTHONHOME &  PYTHONPATH variable

### Anything else?

The last version of Realhunter won't works also ( https://github.com/realthunder/FreeCAD/releases/tag/2022.04.30-edge )
But a previous one works ( FreeCAD-asm3-Stable-Win64-Py3-Qt5-20220111 )

The Stable version of freecad (19.4) working properly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-09:
Please open a discussion first in the forum. This is our platform to discuss. When other in the forum see the issue is real an issue can be opened.

> It's not working also without PYTHONHOME & PYTHONPATH variable

Did you check if you have these variables not for the current user AND for the admin user? Every user can have its own PYTHONHOME while e.g. the admin account has not or vice versa.
Since you can also not start other builds, it is something with your configuration.
If nothing helps, try to uninstall Python, then install FreeCAD.
Alternatively you can also try to uninstall FreeCAD _completely_. This is done by checking in the uninstaller the option to remove also the user preferences.

> I'm on Windows 10 Entreprise version 1803

I think this is no longer supported by Microsoft. Maybe just upgrading Windows might fix the issue for you since also Python and Co. stop at some point the support of unsupported Windows versions.

-> Closing this issue, reopen after forum discussion.

closed 2022-06-09 by donovaly

luzpaz, 2022-07-06:
@DoumeDuSud did you open a forum thread for this ? if so, URL please?