FreeCAD/FreeCAD #7005: [Bug] Python script export STL file is damaged

lindow2009, 2022-06-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=40&t=69325

### Version

0.19 (Release)

### Full version info

```shell
0.18.2
```

### Subproject(s) affected?

Part

### Issue description

environment：centos7
version：0.18.2

python code：
import FreeCAD
import Part
import Mesh
shape = Part.Shape()
shape.read('my_shape.step')
doc = App.newDocument('Doc')
pf = doc.addObject("Part::Feature","MyShape")
pf.Shape = shape
Mesh.export([pf], 'my_shape.stl')

However, the exported STL file and step source file have missing data and incomplete graphic display. Have you encountered the same problem and how to solve it?

Thank you very much.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-11:
There is no info given how to reproduce the bug. (no example file, no step by step instructions).
Please only create an issue when others in the forum can reproduce the bug.

And you are using the outdated version 0.18 of FreeCAD. Please upgrade.

closed 2022-06-11 by donovaly

luzpaz, 2022-06-11:
@lindow2009 CentOS is usually pretty behind when it comes to distro packages. Also 0.18.x is ancient history.    
[image]

If you want to use a more modern version of FreeCAD consider using the Snap, Flatpak, AppImage or PPA.

lindow2009, 2022-06-11:
How to install the latest version of freecad on CentOS ？

lindow2009, 2022-06-11:
@luzpaz

luzpaz, 2022-06-11:
This is better conversation for the forum because we're spamming everyone else watching this repository. But in short
* [Snap](https:///wiki.freecad.org/Snap) (CentOS >= v7.6 per [snapcraft docs](https://snapcraft.io/docs/installing-snap-on-centos))
* [Flatpak](https:///wiki.freecad.org/Flatpak) (CentOS >= v7 per [flatpak docs](https://flatpak.org/setup/CentOS)
* [AppImage](https:///wiki.freecad.org/AppImage) I'm unsure what the status is for CentOS.

lindow2009, 2022-06-11:
@luzpaz I have successfully installed freecad with snap, but I can't call it with Python.

luzpaz, 2022-06-11:
Please open a ticket on https://github.com/FreeCAD/FreeCAD-snap