FreeCAD/FreeCAD #7014: [Feature Request] Allow writing comment / notes inside sketches

akshimassar, 2022-06-13:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?style=1&p=189271

### Subproject(s) affected?

Sketcher

### Idea description

It would be useful to be able to write notes / comments inside a sketch. For instance, I don't always need to have full parametric model. If I change some dimensions in the sketch I would like to be able to leave a note for future myself about reasoning behind that change. Others work that around by using Draft Text or Spreadsheet which is not very convenient.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-19:
You references a year-old forum thread. Please open a new one please.

A general comment tool would be nice, not just one for every workbench. So when e.g. extruding a sketch, one also need to comment on that but this would then be in the Part or PartDesign workbench.

Reopen this issue with the results of a fresh discussion.

closed 2022-06-19 by donovaly

julian-weinert, 2023-09-26:
I definitely need the same thing. It's sometimes necessary to remind you why you did something a certain way, or that you're referencing something else.
It's like comments in software. As soon as you do something unclear / out of the ordinary / weird you add comments to remind yourself and others why you wrote it in some quirky way.

Why would one need to comment on the extrude when you commented something on the sketch? Here is a little mockup of what would be more than enough.

[image]