FreeCAD/FreeCAD #7018: [Bug] 3mf created with FreeCAD cannot be loaded with PrusaSlider

wwmayer, 2022-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=63095

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 18.04.6 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.29165 (Git)
Build type: Debug
Branch: master
Hash: c0cccf7f7ea87663cca813c147f81380c87e6969
Python 3.6.9, Qt 5.9.5, Coin 4.0.0a, Vtk 7.1.1, OCC 7.3.0
Locale: German/Germany (de_DE)
[/code]
```

### Issue description

PrusaSlider or Fusion 360 cannot load an 3mf file created with FreeCAD. However, Blender is able to load it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2022-06-14:
PR: https://github.com/FreeCAD/FreeCAD/pull/7023

closed 2022-06-14 by wwmayer

chennes, 2022-06-14:
Do we want to backport this for 0.20.1?

luzpaz, 2022-06-14:
It makes sense to.

commit 79df90a4 (2022-06-14): Mesh: fixes #7018: 3mf created with FreeCAD cannot be loaded with PrusaSlicer
commit 28f730a8 (2022-06-19): Mesh: fixes #7018: 3mf created with FreeCAD cannot be loaded with PrusaSlicer