FreeCAD/FreeCAD #7020: Files with with names beginning with numbers get clipped in the "Selection view" panel.

gitdocke, 2022-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=601753&sid=fa8ad0ddb637695426c36e1f1409b4de#p601753

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 10 (buster) (KDE//usr/share/xsessions/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.29085 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1e96bbd)
Hash: 1e96bbd2815683d6afa0afe8f985cdd04ef839fe
Python 3.10.4, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * Silk 1.0.0
  * A2plus 0.4.56
  * Assembly4 0.12.0
  * Help 1.0.3
  * Assembly3 0.11.3
  * offline-documentation 1.0.0-alpha
[/code]
```

### Issue description

Files with with names beginning with numbers get clipped in the "Selection view" panel.
Pics 2 and 3 are the same file but saved with different names.

[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-06-17:
@wwmayer any idea what is causing this?

luzpaz, 2022-07-15:
@gitdocke is it just a number at the beginning or does the file name need to be a certain amount of characters long in order for the bug to appear ?

onekk, 2022-07-22:
I think that is caused by the fact that a "filename" could be used internally as a Python variable, and Python variables should have not a number as starting character, sadly I could not find anything precise in the Python.org site, but as many other documentations like w3schools and other python tutorials cite this limitations, somewhere it was told.

Regards

Carlo D.

wwmayer, 2022-07-27:
The file name is used as an identifier and because the first character cannot be a digit (not even in Py3) it will be replaced with an underscore. See also: https://docs.python.org/3/reference/lexical_analysis.html#identifiers:

> Within the ASCII range (U+0001..U+007F), the valid characters for identifiers are the same as in Python 2.x: the uppercase and lowercase letters A through Z, the underscore _ **and, except for the first character, the digits 0 through 9**.

mnesarco, 2023-03-28:
Why not prepend the underscore instead of replacing the first char during filename sanitizing?

sliptonic, 2023-05-01:
This issue is still tagged for a 0.20.1 release milestone.   
I'm going to remove the milestone.  

Is there anything to do on this issue?  Can it be closed?

adrianinsaval, 2023-05-08:
@wwmayer what's your opinion on mnesarco's proposal of prepending the underscore instead of replacing the digit?

Beyond that I don't think there's much else we can do here.

chennes, 2023-05-20:
The rules for filenames are unrelated to the rules for identifiers (and vary based on the OS!) -- can we stop using the filename as an identifier? Replace it with a hash, or a UUID, or something? I can't find the specific line(s) of code where this is happening, can someone point me in the right direction?

wwmayer, 2023-05-22:
https://github.com/FreeCAD/FreeCAD/blob/master/src/App/Application.cpp#L588

adrianinsaval, 2023-05-22:
> The rules for filenames are unrelated to the rules for identifiers (and vary based on the OS!) -- can we stop using the filename as an identifier? Replace it with a hash, or a UUID, or something? I [...]

but then it'll be even more confusing to see some random string in the selection view with no obvious connection to the file in the user's eyes, plus for hash and UUID we would also have to replace leading numbers with underscores.

maxwxyz, 2024-07-27:
This is still an issue in the latest weekly:
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

mnesarco, 2025-09-04:
This seems to be fixed some time ago:
https://github.com/FreeCAD/FreeCAD/blob/0020057f8383a023679cddb1f037cda7deda2fa8/src/Base/Tools.cpp#L82