FreeCAD/FreeCAD #7030: [Feature Request] Adjust constraint indicators to the size of the model

wwmayer, 2022-06-15:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=18&t=69445

### Subproject(s) affected?

FEM

### Idea description

For very small models a hard-coded size of the constraint indicators causes the problem that in the view they appear as huge object and cover the model. This makes the further handling very difficult.

What's needed is a way to make the size of the indicators depending on the size of the model.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NXTler, 2023-08-26:
I want this feature too

luzpaz, 2024-03-09:
@FEA-eng is this on the FEM roadmap ?

FEA-eng, 2024-03-09:
> @FEA-eng is this on the FEM roadmap ?

No, because solving this one should be sufficient: https://github.com/FreeCAD/FreeCAD/issues/11306

There is some size dependence already but it's not working properly and we need a functional Scale property for those symbols.

PR #13586 (2024-04-29): Fem: Constraint symbol scaling based on shape size - fixes #7030
To be merged after https://github.com/FreeCAD/FreeCAD/pull/13274.
@FEA-eng could you test? The last commit fixes #7030 (initial scaling for very small and big objects).
Scaling is based on a characteristic length of the objects, basically the diagonal of the bounding box, so symbols scale linearly with the shapes.
commit ba0825f6 (2024-04-29): Fem: Constraint symbol scaling based on shape size - fixes #7030
closed 2024-04-29 by chennes