FreeCAD/FreeCAD #7042: [Bug] Can't import FreeCAD in python. DLL load failed while importing FreeCAD: The parameter is incorrect.

rydb, 2022-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=602809#p602809

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
```

### Issue description

BUG: When attempting to import FreeCAD in python, I get the error, "ImportError: DLL load failed while importing FreeCAD: The parameter is incorrect."

Steps to reproduce:

create a virtual environment with python 3.8:
[https://www.python.org/downloads/release/python-380/](https://www.python.org/downloads/release/python-380/)

`py -m virtualenv -p="<installed python 3.8 exe location>" .virtenv`

import FreeCAD with the following script:

```
FREECADPATH = '/program files/FreeCAD 0.20/bin' # path to your FreeCAD.so or FreeCAD.dll file
import sys
import os
sys.path.append(FREECADPATH)

#print(sys.version)

import FreeCAD

print("free cad is " + str(FreeCAD))
```

Run the above script with the python 3.8 of the virtual environment set to your interpreter.
The above script will result in the error:

`DLL load failed while importing FreeCAD: The parameter is incorrect. `

at least it will for versions 0.19 and 0.20 of FreeCAD(I've only tested those versions)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-06-17:
@donovaly ping

donovaly, 2022-06-18:
> @donovaly ping

What should I do? Looking in the thread I don't see that e.g. a Python file is missing. I am no Pythionist but to me the forum thread reads this is no bug.

donovaly, 2022-06-19:
Closing for now. When there is a consensus that there is a bug and what should/must be changed, please reopen.

closed 2022-06-19 by donovaly