FreeCAD/FreeCAD #7043: [Bug] Model view distorted when Draft grid is enabled

FEA-eng, 2022-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=23&t=69527

### Version

0.20 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Draft

### Issue description

I was trying to move Draft grid to the datum plane. When I did this, the model view was distorted. It looks similar to this known issue: 

https://forum.freecadweb.org/viewtopic.php?f=3&t=64833

but it doesn't disappear after pressing Fit All. Only hiding the grid helps.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2022-06-28:
IMO this issue can be closed.

luzpaz, 2022-06-29:
Closed by [657f049](https://github.com/FreeCAD/FreeCAD/commit/657f04908f6ef4184249b25db79951836ede8afd) in #7105