FreeCAD/FreeCAD #7048: [Bug] Sketcher Carbon Copy does not work if source sketch is attached to target sketch

derMart, 2022-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=44&t=69561&p=603361#p603361

### Version

0.20 (Development)

### Full version info

```shell
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.617.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.06.17-edge)
Hash: 22643458936e013c38a8d7069aede81b5bbe42fa
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.2
Locale: English/United States (en_US)
```

### Issue description

### Steps to reproduce

- create a sketch "Sketch"
- create another sketch "Sketch001" which is attached (Attachment->Support) to "Sketch"
- create a circle in "Sketch001"
- go to "Sketch" and try to carbon copy the contents of "Sketch001" into "Sketch"

### Actual result
carbon copy is not allowed, stating that there is a circular dependency problem

### Expected result
carbon copy operation should work as it should copy the geometry contents of "Sketch001" into "Sketch". This should be the same outcome as to manually create the circle in "Sketch". This outcome is independent of the fact, that "Sketch001" is attached to "Sketch".

### Anything else?

Documentation of the carbon copy tool: https://wiki.freecadweb.org/Sketcher_CarbonCopy does not state a limitation where the source can't be attached to the target sketch.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-19:
@abdullahtahiriyo , reading the linked forum I tend to say this is no bug. Can you please have a look?

derMart, 2022-06-19:
@donovaly I strongly disagree, have you read the thread? Please have a look at my latest post: https://forum.freecadweb.org/viewtopic.php?p=603513#p603513
The bug is even more generic. As proven in the post, carbon copy theoretically never should create a circular dependency (between constraints). Constraint dependencies are the only thing which sould prevent an operation. That a sketch is attached to another sketch is even more independent from the fact that any circular dependencies may occur after carbon copy.

abdullahtahiriyo, 2022-10-15:
@0penBrain assessment is good. There is no bug. There is room for improvement.

Could be reclassify this ticker as "Feature Request"?

donovaly, 2022-10-15:
> could be reclassify this ticker as "Feature Request"?

Done.

abdullahtahiriyo, 2022-10-16:
@donovaly Dankeschön.

adrianinsaval, 2023-04-21:
Rather than a carbon copy problem I think this is a general issue with how FreeCAD checks for dependencies, do we have an existing issue for this general problem?

adrianinsaval, 2023-05-11:
#8059 is the closest to a more generalized issue (actually it talked of spreadsheets but is easier to reword into a general one IMO), shall we close this in favor of that one?