FreeCAD/FreeCAD #7050: [Bug] <title> Stairs number of steps & still touched...

bgma, 2022-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=23&t=68424

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 7 SP 1 (6.1)
Word size of FreeCAD: 64-bit
Version: 0.20.29182 (Git)
Build type: Release
Branch: master
Hash: e89ebba7369dca5775fd8aa878a0d3c494e6b2ac
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: Swedish/Sweden (sv_SE)
[/code]
```

### Subproject(s) affected?

Arch

### Issue description

I've seen some issues with ArchWB / BIM: Stairs
When I enter ex. 11 steps are created 10 steps.  and railings
When I enter 2 steps, no steps are created and railings
When I enter 1 step, no steps are created but a plate with "Structure Thickness" and railings

16:31:47  0 <App> Document.cpp(3546): Unnamed#Rail still touched after recompute
16:31:47  3.0343e-05 <App> Document.cpp(3546): Unnamed#Rail001 still touched after recompute

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-19:
@yorikvanhavre might want to have a look.

yorikvanhavre, 2022-06-20:
I'll have a look...
For the numbering, that's usually the logical thing, when you say a stair has two steps, it means two times go "up". the stairs object therefore has only one "block", because the last step is the floor you are reaching. But there could be an option to change that behaviour, if there is some need for it

yorikvanhavre, 2022-06-20:
@paullee0 might want to look too

paullee0, 2022-06-20:
Seems to be something I long time ago created when started learning python.  By the way, there had been several discussion about the improvement of Arch Stairs; hope those can be handled soon.

Roy-043, 2023-02-12:
#8444 does not fix the stairs number issue which is not a bug.

bgma, 2023-02-15:
> #8444 does not fix the stairs number issue but that is not a bug.

Don't understand the reasoning that it's not a bug?
Discovered today that nothing really happens at all when I enter: Number Of Steps= 2, Should give 1 visible step? if 3 Gives 2 Steps?

paullee0, 2023-02-15:
I see what @bgma means now :-
- particularly at when Numbers Of Steps = 2, should return a stair of 1 step?
- and when Numbers Of Steps = 1, should return something ?

- (when there is a Base, e.g. a Draft line, when Numbers Of Steps = 1, it return a slab with same level - it can work as the 'multi-segment' landing I created some time ago)

Roy-043, 2023-02-16:
Sorry, for closing prematurely. I'll have another look.

The steps=2 case should probably work. Steps=1 is problematic given the current behavior where n steps result in n-1 treads.

@paullee0 This touches on a fundamental issue we see with many Arch objects. Certain inputs will either not create a shape or the existing shape will not update.

paullee0, 2023-02-16:
Thanks Roy-043.

Step = 2 seems should work, maybe I have a look when there is a gap.

For Step = 1, see the particular usecase and discussion thread - the updated behaviour was not included in the wiki

There are cases Arch Objects just remains silent, there should be at least some warning at Report pane - not sure if there is any for the concerned case.  Better with a pop-up dialog ?

Roy-043, 2023-02-16:
Bug I found while testing: #8514

PR #8444 (2023-02-13): [Arch] fix Arch_Stairs recompute issue
Fixes #7050.

The recompute issue was due to the inclusion of the railings in the `Additions` property of the stairs object. The `RailingLeft` and `RailingRight` properties now have the `"App::PropertyLinkHidden"` type and the referenced objects are claimed as children.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---
PR #8515 (2023-02-16): [Arch] Allow Arch_stairs with 2 steps
- Fixes #7050

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---
commit dfc57aa8 (2023-02-16): [Arch] Allow Arch_stairs with 2 steps (#8515)

- Fixes #7050
commit c689fc1f (2023-02-17): [Arch] Allow Arch_stairs with 2 steps (#8515)

- Fixes #7050
closed 2023-02-16 by Roy-043