FreeCAD/FreeCAD #7052: [Bug] <title>

quiret, 2022-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=69542

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/Germany (en_DE)
[/code]
```

### Issue description

https://www.youtube.com/watch?v=gWf-_U64IMo

When I attach a shape binder which has a selection of geometry using the Data tab of the Combo View then the selection set gets overwritten along with the attachment Support.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-19:
@realthunder , can you please have a look at this?

quiret, 2022-06-23:
Please note this aswell: https://forum.freecadweb.org/viewtopic.php?p=604704#p604704

jcoalson, 2024-02-17:
I added a simple example to reproduce here: https://forum.freecad.org/viewtopic.php?p=741301#p741301

Everything appears to work until the project is saved.

jcoalson, 2024-02-23:
Another discussion: https://forum.freecad.org/viewtopic.php?t=76445

commit a8ae56e0 (2024-03-04): Part: Rename AttachExtension::Support property to AttachmentSupport, to avoid name conflict with base features.  Fixes #7052
closed 2024-03-04 by adrianinsaval