FreeCAD/FreeCAD #7054: [Bug] Draft to Sketch issue

bgma, 2022-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=69508

### Version

0.20 (Development)

### Full version info

```shell
[code]
OS: Windows 7 SP 1 (6.1)
Word size of FreeCAD: 64-bit
Version: 0.20.29192 (Git)
Build type: Release
Branch: master
Hash: fb1a7af3605ed86631afabb1eb77ae305e418d98
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: Swedish/Sweden (sv_SE)
[/code]
```

### Subproject(s) affected?

Draft

### Issue description

Hello,

I saw something that I think is a little strange when
I was testing the Draft to Sketch function.

Part design WB
Create Body
Create Sketch
XY-Plane
Create Rectangle over the cross
[Close]
Make Pad
Select Front
Draft WB
Circle
Select Circle
Draft to Sketch
Select Bottom

The Circle has been created at origin line
Is this as it should be?

What is the reasonableness of it being created in a different plane than the one where the original was? I think it's a BUG

/Best Regards BG 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-19:
Can you please give it a better description (title and less steps to reproduce)?

Also please attach a test file to your forum post. that one can focus on the steps to actually reproduce the issue?

bgma, 2022-06-19:
The draft circle is created on the working plane while the sketch is created on the main plane

[image]
[image]

[pad-3.zip](https://github.com/FreeCAD/FreeCAD/files/8936134/pad-3.zip)

donovaly, 2022-06-19:
Closing in favor of #7058

closed 2022-06-19 by donovaly