FreeCAD/FreeCAD #7058: [Bug] [Draft] Draft to sketch does not take placement into account

donovaly, 2022-06-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=603648#p603648

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
[/code]
```

### Subproject(s) affected?

Draft

### Issue description

- take this example document: 
[Draft-to-Sketch.zip](https://github.com/FreeCAD/FreeCAD/files/8936554/Draft-to-Sketch.zip)
- select the circle object in the tree view
- use the command Draft to Sketch from the Draft WB

result: the placement of the circle object was ignored. The resulting sketch has a placement of (0, 0, 0).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-19:
@Roy-043 I am not sure if this behavior might be by design. Can you please have a look? The Wiki page does not state anything in this regard: https://wiki.freecadweb.org/Draft_Draft2Sketch

Roy-043, 2022-06-22:
Using the Placement of the Draft objects would not always work. For example, a planar Draft Wire need not lie on the XY plane of the coordinate system defined by its Placement. Just create a triangular Draft Wire, edit its Points property and change the Z of one of the points. The issue can be be solved by changing the base of the sketch. I'll do a PR.

Roy-043, 2022-06-22:
Note that there is another, separate, issue: The Draft_Draft2Sketch command does not take the Placement of parent objects into account.

Roy-043, 2022-06-28:
IMO this issue can be closed.

closed 2022-06-30 by luzpaz