FreeCAD/FreeCAD #7059: [Bug] User edit modes are broken for a long time,

MisterMakerNL, 2022-06-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=69579

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly3 0.11.3
  * Behave-Dark-Colors 0.0.1
  * BIM 2021.12.0
  * FeedsAndSpeeds 0.4.0
  * FEM_FrontISTR 0.1.0
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0
  * SelectorToolbar
  * sheetmetal 0.2.49
  * ToolbarStyle
```

### Issue description

I was trying features too design Icons for and came across the user edit modes. 
My experience was:

- Transform mode
 Doesn't work on bodies and is basically a 1 click less to open the transform feature. (right click transform)
Although it does work and would maybe the only button that would be useful to keep. 
- Paint mode
 Just starts the paint faces feature but doesn't select any faces of any feature you clicked on.
- Cut mode
 Does nothing???

I did some reading on  [github](https://github.com/FreeCAD/FreeCAD/pull/5265)  and there was also some discussion before about how broken it is, but that was 1 year ago and it is still broken. It was brought up that maybe an add-on was using it and so would not be removed. If an add-on is using it can't we let the add-on make a button? And keep the code in the backbone, but out of the toolbar? It's pretty bad if you have icons in the main toolbar that do absolutely nothing.

My suggestion is just remove the buttons from the toolbar completely if anyone is using it they will complain and we can discuss a better way of implementing the buttons without the non-working features. If an add-on is using it let the add-on make a button in a toolbar that calls this function.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2022-06-20:
Please first open an issue if users in the forum agrees there is one. Just requesting to remove a brand new feature rather than at least contacting the developer who implemented it and give him a change to improve it, is not fair (to be diplomatic).

closed 2022-06-20 by donovaly

MisterMakerNL, 2022-06-20:
It was just a suggestion!
I'm not diplomatic, if you couldn't tell. ;)