FreeCAD/FreeCAD #7062: [Feature Request] Draft label with hyperlink

PeterPNoster, 2022-06-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=23&t=69528

### Subproject(s) affected?

Draft

### Idea description

Is it possible to set an draft label on the 3D Part and set up an hyperlink, to an internal/external document, such as a pdf drawing or a text document?
Click on the draft label and the system viewer displays the document.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2022-06-21:
Clicking an object and immediately open an URL is IMHO delicate, as people will constantly pop up browsers when working...
What we could do is

1. Detect when a label text contains an URL (begins with http, basically...)
2. If yes, add a context menu option to open the URL

> It would be great if such a function could be exported and used as well in a n .xhtml file.

I didn't understand that?

yorikvanhavre, 2022-06-21:
Oh I see, you mean the webGL export. I think we still need to support 2D objects and texts there, but will have a look.

luzpaz, 2022-07-09:
@PeterPNoster than can you update the ticket summary to indicate hyperlink for webGL export perhaps?

PeterPNoster, 2022-07-09:
> @PeterPNoster than can you update the ticket summary to indicate hyperlink for webGL export perhaps?

@luzpaz I've updated the summary.

luzpaz, 2023-09-04:
Anyone interested in working on this ?

luzpaz, 2023-09-14:
@CalligaroV interested to take a stab at this ?

CalligaroV, 2023-09-14:
@luzpaz I think I can at least try :wink: 
Speaking the truth ATM I was preparing a couple of pull requests for https://forum.freecad.org/viewtopic.php?t=80253 and https://forum.freecad.org/viewtopic.php?t=80696 but I think I can use some time for this issue too in the next days.

However ATM I can't say if I'll be able to come out with something useful and how much time I could need, I haven't even started reading the related forum topic yet :sweat_smile:

luzpaz, 2023-09-14:
@CalligaroV all good. No rush. Just putting it on your radar.

yorikvanhavre, 2023-09-18:
What I would do:

# Create a function in Draft that finds, extracts and returns an URL from a piece of text (there are certainly thousands of regexes to do that ready on the net)
# Create a new Draft command that picks the URL of a selected text, and opens a browser to it
# In the Draft's Workbench code (IntiGui.py), in the context menu setup, if the selection contains just one Draft text and that texts contains an URL, display the command

For the webGL export I'm not sure, more research is needed there...

CalligaroV, 2023-10-01:
Hi Everyone! :smiley: 

After some research, coding and tests, I just opened a pull request for the modifications needed to use the hyperlinks in Draft WB, following @yorikvanhavre tips.
you can see above the link, Pull Request #10878.

How it works:
- the command allows the selection of multiple object, even multiple texts.
- for every text a regex will be applied. If at least one text contains at least one hyperlink the command will be added to the context menu.
- when called the command, in case of multiple hyperlinks a message box will inform that multiple windows could open (one for every hyperlink)
- for every hyperlink a subprocess will launch the application related to the file type / protocol and open the file/page described by the hyperlink

Some notes:
- the regex accepts spaces in the hyperlink only before the first dot, which is supposed to be the beginning of the file extension. Issues may arise if there are combinations of dots and spases in the path of the file or in the filename itself (for example /home/vincenzo.calligaro/some folder with spaces/file.txt will fail due to the dot in the folder vincenzo.calligaro and the following spaces in the folder "some folder with spaces")
- the code should already be platform independent and support Windows, macOS and Linux but ATM **I tested it only on Linux** (Linux Mint 12 Cinnamon)
- the code to actually open the files in the hyperlinks, launching the correct application, is more or less what I found here: https://stackoverflow.com/questions/17317219/is-there-an-platform-independent-equivalent-of-os-startfile

Final note: haven't done any research for the WebGL

As always, feedbacks are very wellcome! :wink:

PR #10878 (2023-10-23): [Feature Request] [Draft] [UI/UX] new Draft_Hyperlink command
Fixes #7062

**Forum discussion**

https://forum.freecad.org/viewtopic.php?f=23&t=69528
closed 2023-10-23 by yorikvanhavre