FreeCAD/FreeCAD #7066: Image scaling in Freecad 2.0 techdraw broken

Tommatous, 2022-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=69656

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.56a
  * fasteners 0.3.45
  * sheetmetal 0.2.49
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

Image scaling broken on tech draw workbench in version 2.0

Works fine in version 0/19

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2022-06-30:
Assuming the OP is talking about TechDraw_Image, this isn't a regression.  It doesn't scale properly in v0.19.

We didn't get any response on questioning the OP, but I suspect that he is actually complaining about regular views.  v0.19 was broken and allowed Scale changes when it shouldn't have.  v0.20 handles Scale and ScaleType properly.

luzpaz, 2022-06-30:
@WandererFan feel free to alter or close this issue then

xsuper9988, 2022-07-02:
Originally thought it was a problem with the Link object
It was later found to be a Scale mirror
Finally found that the Scale size is also a problem
Is this problem the same?
#7113

WandererFan, 2022-07-04:
should be fixed by commit 11c6f4a9e9

closed 2022-07-04 by WandererFan

WandererFan, 2022-10-11:
I shouldn't think so.  #7066 is about scaling a bitmap image on a TechDraw
page.  #7113 seems to involve reversed normals in the 3d view.