FreeCAD/FreeCAD #7070: [Bug] Hang on start up when using dual monitors

e12voltsdac, 2022-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?style=3&f=4&t=64973

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus
  * Assembly4
[/code]
```

### Issue description

FreeCAD will hang on start up when I have last used the program on my second monitor. It seems to be related to the last window position and trying to start from there. Opening FreeCAD results in a frozen blank window, need to kill from Task Manager.

If I modify the registry key for window position to 'reset' it, then I can successfully start FreeCAD. Another work around is disabling the second monitor to force FreeCAD to open on my native monitor.

On Windows 10, monitor resolutions are both 2560 x 1440. Windows scale for viewing set to 100% (no scale).

I can reproduce the issue each time I start FreeCAD by closing it from my second monitor.

Registry key: 
[HKEY_CURRENT_USER\SOFTWARE\FreeCAD\FreeCAD\Qt5.15]
"Position"="@Point(-1 -8)"

### Anything else?

Issue is mentioned in the following forums:

https://forum.freecadweb.org/viewtopic.php?style=3&f=4&t=64973

https://forum.freecadweb.org/viewtopic.php?style=4&p=599231

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-06-24:
Sounds like a Qt related multimonitor issue. Any upstream tickets open regarding this?

adrianinsaval, 2022-06-25:
https://forum.freecadweb.org/viewtopic.php?f=3&t=69071
https://forum.freecadweb.org/viewtopic.php?f=4&t=69664
other people also had this problem

adrianinsaval, 2022-06-25:
@wwmayer does freecad do anything to restore the window position of previous session or is this all auto managed by qt?

wwmayer, 2022-06-26:
Yes, it saves current window size, position and the layout of the toolbars or dock windows. Depending on the OS the information is saved in different locations. E.g. on Windows it's saved to the registry or under Linux to the file FreeCAD.conf.

chennes, 2022-06-26:
For the record this is working fine for me under Windows 11, except that FreeCAD always launches on my primary display, regardless of where I had it last.

e12voltsdac, 2023-02-11:
Just wanted to follow up on this issue. I downloaded 20.2 and the issue still exists. Looks like the Qt version is still the same. 
Qt 5.15.2

Does this need to be escalated up to a Qt ticket? 
Thanks!

chennes, 2023-02-11:
No point - they don't support 5.15 anymore, so I certainly wouldn't expect any action on it. I wonder if we can detect an invalid window position and reset it ourselves -- @wwmayer do you know if it's possible to "intercept" or otherwise detect Qt's window reloading code?

wwmayer, 2023-02-15:
The restoring of the main window is done here: https://github.com/FreeCAD/FreeCAD/blob/master/src/Gui/MainWindow.cpp#L1463

luzpaz, 2023-03-15:
bumping for visibility

luzpaz, 2023-09-04:
Anyone want to pick up the mantle on this issue?

adrianinsaval, 2023-09-19:
I can't reproduce this on windows 11
```yml
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34089 +13 (Git)
Build type: Release
Branch: pr/Pesc0/9663
Hash: ed74312cd933b0f80ca4010460f53598872cd641
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: C/Default (C) [ OS: Spanish/Paraguay (es_PY) ]
Installed mods: 

```

luzpaz, 2023-09-23:
@e12voltsdac can you reproduce on development build ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

luzpaz, 2023-11-01:
Another dual monitor issue: #11102

luzpaz, 2023-11-17:
@macdroid53 @maxwxyz can you reproduce ?

macdroid53, 2023-11-17:
I have started today's code on Debian/XFCE and Manjaro/Plasma, with dual and triple monitors, no issue.

luzpaz, 2023-11-17:
Can anyone test on windows ?

adrianinsaval, 2023-11-17:
No issue on windows 11

luzpaz, 2023-11-17:
@e12voltsdac we're going to close this ticket. If you can reproduce on latest stable 0.21.2 or latest development 0.22.x please re-open.

closed 2023-11-17 by luzpaz