FreeCAD/FreeCAD #7078: [Feature Request] reduce/eliminate draft wb unnecessary output messages

adrianinsaval, 2022-06-23:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=604611#p604611

### Subproject(s) affected?

Draft

### Idea description

Draft WB tools output a lot of unnecessary messages to the report view, for example:
> 19:55:27 ----------------
> 19:55:27 GuiCommand: Line
> 19:55:27 Pick first point
> 19:55:32 Pick next point

these should be removed since they don't serve any meaningful purpose or at most sent to the log instead

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2022-06-23:
Indeed I would also like to turn all this off...

luzpaz, 2023-11-18:
Something we should tackle before 1.0, right ?

yorikvanhavre, 2023-11-20:
Definitely.

The problem is that those are hard-coded throughout Draft using _msg calls

I think the best course of action would be to add a new function in Draft/utils/messages, ex. _toolmsg which would only print the message depending on a preference option, so they can be turned on/off by the user.

adrianinsaval, 2023-11-21:
is there any use case for seeing these messages?

yorikvanhavre, 2023-11-23:
It started as an experiment to replicate the old autocad workflow, where you had a "command window" where you could do everything with the keyboard, in a kind of text-based interactive interface. It was really fast.

It was carried a bit further afterwards but never completely finished... I think someone might want to resurrect this and work further on it at some point, so I wouldn't throw it all away... But yes in its current form it has basically no real use.

PR #11503 (2023-11-24): Draft: Added a preference to turn on/off tool messages
New property: Draft -> General -> Show tool messages (off by default)
Fixes #7078
commit b1bcfec2 (2023-11-24): Draft: Added a preference to turn on/off tool messages (off by default) - fixes #7078
closed 2023-11-24 by Roy-043