FreeCAD/FreeCAD #7079: [Bug] AddOn manager fails to initialise, 0.21 daily build on ubuntu ppa

Videonauth, 2022-06-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=69715

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 20.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.
Build type: Release
Branch: unknown
Hash: a527daa4b3341688073ed7a8dee107e80109276d
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.56a
  * Render 2022.2.0
  * FeedsAndSpeeds 0.4.0
  * BOLTSFC
  * Assembly4 0.12.0
  * 3D_Printing_Tools
  * fasteners 0.3.45
  * parts_library
  * Curves 0.4.4
  * Help 1.0.3
  * Behave-Dark-Colors 0.0.1
  * fcgear 1.0.0
```

### Subproject(s) affected?

Addon Manager

### Issue description

The Addon manager fails to initialize on start up on the current daily-build and did so with the previous days one as well, producing following error message:

```
10:51:21  Init:      Initializing /usr/share/freecad-daily/Mod/AddonManager... failed
10:51:21  ----------------------------------------------------------------------------------------------------
10:51:21  Traceback (most recent call last):
  File "<string>", line 140, in RunInitGuiPy
  File "<string>", line 6, in <module>
  File "/usr/share/freecad-daily/Mod/AddonManager/AddonManager.py", line 38, in <module>
    from addonmanager_workers import *
  File "/usr/share/freecad-daily/Mod/AddonManager/addonmanager_workers.py", line 84, in <module>
    import markdown
  File "/home/michael/.local/share/FreeCAD/AdditionalPythonPackages/markdown/__init__.py", line 29, in <module>
    from .core import Markdown, markdown, markdownFromFile  # noqa: E402
  File "/home/michael/.local/share/FreeCAD/AdditionalPythonPackages/markdown/core.py", line 26, in <module>
    from . import util
  File "/home/michael/.local/share/FreeCAD/AdditionalPythonPackages/markdown/util.py", line 88, in <module>
    INSTALLED_EXTENSIONS = metadata.entry_points(group='markdown.extensions')
TypeError: entry_points() got an unexpected keyword argument 'group'
10:51:21  ----------------------------------------------------------------------------------------------------
10:51:21  During initialization the error "entry_points() got an unexpected keyword argument 'group'" occurred in /usr/share/freecad-daily/Mod/AddonManager/InitGui.py
10:51:21  Please look into the log file for further information
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Videonauth, 2022-06-23:
So after the forum discussion it seems, the version of markdown packaged with the daily build on the ppa is not sufficient to run the Add-on manger, installing markdown system wide with pip did solve the above error message.

luzpaz, 2022-06-24:
@yorikvanhavre we need a PPA maintainer. any ideas who we can nominate?

yorikvanhavre, 2022-06-24:
Is markdown really packaged in the ppa version? That's strange...
Question: With snap packages now kind of "mainstream" in ubuntu, does it still mae sense to provide a ppa?
Otherwise,I have not much idea of who could fill the slot...

Videonauth, 2022-06-24:
Oh please, do not get rid of the ppa, there are so many people in the Ubuntu realm who despise snap as it is. This is my Observation of being Admin on one of the largest Ubuntu Community servers.

chennes, 2022-06-24:
Really for me the easy fix is to remove the checks for Markdown: it is basically obsolete in the current Addon Manager, it's not used by the mainline code anymore. I was only keeping it around as a fallback.

luzpaz, 2022-06-25:
I though that was the way Addon Manager displays markdown content from github within FC ?

Thanks chennes for jumping in. Still the point is that we need a PPA maintainer. @Videonauth know anyone. Can you put the word out?

Videonauth, 2022-06-25:
I can ask around in the Ubuntu community, but isn't anyone of the ppa team members existent still?

luzpaz, 2022-06-25:
None of those folks are available unfortunately, AFAIK

Videonauth, 2022-06-25:
So, what are the requirements you would have a possible maintainer to fulfill to be considered, so we on the  discord server can ask possible candidates before referring them here?

luzpaz, 2022-06-25:
Hi @sgrogan, @kkremitzki would one of y'all share what the job or a PPA maintainer looks like ?

WandererFan, 2022-07-16:
>would one of y'all share what the job or a PPA maintainer looks like ?<

Late to the party, but this might help: https://www.forum.freecadweb.org/viewtopic.php?f=42&t=69858

luzpaz, 2022-08-05:
Still haven't found a maintainer which is stalling this issue.

luzpaz, 2022-09-12:
@kkremitzki are you back to helping maintain the PPA ?

luzpaz, 2023-01-09:
We still don't have active maintainer(s) for the PPA

luzpaz, 2023-03-08:
@kkremitzki do you happen to know if there is any traction on finding a maintainer for PPA ?

yorikvanhavre, 2023-03-08:
There is some interesting development at https://forum.freecad.org/viewtopic.php?p=665058#p665058

luzpaz, 2023-03-15:
@yorikvanhavre still no love :disappointed:

kkremitzki, 2023-03-16:
I saw it, but haven't had a chance to look at it deeply and write up a reply. My first impression was that, in general, the approach is what I'd like, but care has to be taken to make sure that existing users will get an update without intervention needed, and that some existing packaging decisions have to be worked around, for example the use of package epochs.

luzpaz, 2023-11-26:
@chennes whats the verdict on the PPA. Can we somehow let users know it's not maintained and need a maintainer via the PPA README?

adrianinsaval, 2024-01-26:
ppa has been updated and is now working

closed 2024-01-26 by adrianinsaval