FreeCAD/FreeCAD #7085: [Bug] <sheetmetal workbench overrides part transparency and colour>

thegavs, 2022-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=69726

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Linux Mint 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

Any action in sheet metal workbench overrides the transparency and colour set in the part.

To replicate create a flat part in part design workbench and set the transparency to a high value I used 80%
then swap to sheet metal workbench and add a bend to a face.
The transparency setting seems then to be ignored but not reset. changing the value will bring the transparency back.

### Anything else?

[image]
[image]
[image]
[sheetmetal transparency.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/8976081/sheetmetal.transparency.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-05:
Related https://github.com/shaise/FreeCAD_SheetMetal/issues/206

luzpaz, 2022-08-05:
Hi @Syres916,  can we close this ticket then ?

Syres916, 2022-08-05:
If @thegavs is happy that individual face colours will have to wait until after TopoNaming is sorted then yes but I was giving a little time for feedback, it's holiday time in the northern hemisphere after all.

thegavs, 2022-08-05:
am happy with that as there is an easy workaround for the time being.

thank you

Syres916, 2022-08-10:
> Hi @Syres916, can we close this ticket then ?

@luzpaz This can now be closed, TIA

closed 2022-08-10 by luzpaz