FreeCAD/FreeCAD #7086: [Feature Request] Parametric projection

CsatiZoltan, 2022-06-24:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=22&t=69732

### Subproject(s) affected?

Part

### Idea description

Projecting a curve onto a surface is available in the `Part` workbench of FreeCAD. However, it is not exposed to the Python API. Therefore, I rely on *PythonOCC* to use the underlying projection implemented in *OpenCASCADE*, see the [documentation](https://dev.opencascade.org/doc/refman/html/class_b_rep_proj___projection.html). The Python code can be found [here](https://gist.github.com/CsatiZoltan/f4fd10bf20923143ba0e0678ea1d3d61).
First, import the following:
```python
import Part
from OCC.Core.BRepProj import BRepProj_Projection
from OCC.Core.gp import gp_Dir
```

This worked well, until PythonOCC was removed from FreeCAD. Possible solutions:
- integrate my solution into FreeCAD so that PythonOCC is not needed any more for this functionality
- bundle PythonOCC with (the conda-packaged) FreeCAD

### Anything else?

Related forum entry: https://forum.freecadweb.org/viewtopic.php?f=9&t=33700&start=20#p529942

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-06-25:
So, IOW, you're asking to expose OCC's  `BRepProj_Projection` class to FC's Python API ?

CsatiZoltan, 2022-06-25:
@luzpaz Yes.

luzpaz, 2022-06-25:
@wwmayer do you mind weighing-in ? TIA!

wwmayer, 2022-06-26:
BRepProj_Projection is already available in our Python API. The methods are part of Part.Shape and they are called `makeParallelProjection` and `makePerspectiveProjection`

CsatiZoltan, 2022-06-26:
@wwmayer Thanks, I did not know about this method.
https://github.com/FreeCAD/FreeCAD/blob/93525ff7eb28418e0237b0a866f797c74adfa569/src/Mod/Part/App/TopoShapePyImp.cpp#L2018

This solves my issue, so I close it. If you want to introduce the parametric projection into FreeCAD, feel free to use the referenced code.

closed 2022-06-26 by CsatiZoltan

luzpaz, 2022-06-27:
@CsatiZoltan where in the documentation would you say this would be helpful for others to find?

CsatiZoltan, 2022-06-27:
@luzpaz I would place a link at the end of the [wiki page](https://wiki.freecadweb.org/Part_ProjectionOnSurface), mentioning when my custom class is useful (written in the *Notes* section of the class docstring). I will modify that GitHub Gist to use `makeParallelProjection` instead of the PythonOCC function, making it compatible with FreeCAD 0.20.

luzpaz, 2022-06-27:
@CsatiZoltan would you like access to the wiki to add it yourself?

CsatiZoltan, 2022-06-27:
@luzpaz Yes, please. That would allow me to start contributing to FreeCAD.

luzpaz, 2022-06-27:
You can [DM me](https://forum.freecad.org/ucp.php?i=pm&mode=compose&u=12229) a valid email address

CsatiZoltan, 2022-06-28:
Added to the wiki.