FreeCAD/FreeCAD #7102: [Bug] [regression] [FEM] crash on canceling point flter dialog

donovaly, 2022-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

.

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.29323 (Git)
Build type: Release
Branch: master
Hash: 8acfd5bce7292535ffa049731ca0df21780951f4
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
```

### Subproject(s) affected?

FEM

### Issue description

- in the attached file:
open the point filter
- change the value for the x coordinate to something of your choice
- leave the dialog with the Cancel button
result: maybe an infinite loop, however, for me Windows steps in and kills the unresponsive FreeCAD.

@wwmayer, since this crash does not occur with 0.20 but in current master, can you please have a look since you are the only one who modified FEM after the release. (I reverted the only commit I made to FEM and the problem persists.)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2022-06-27:
> in the attached file:

Which attached file?

donovaly, 2022-06-27:
> Which attached file?

Oops, now I attached it to the initial report as ZIP (is too large for the forum).

wwmayer, 2022-06-27:
No problem here. The dialog disappears immediately and the application doesn't become unresponsive.

donovaly, 2022-06-27:
> No problem here. The dialog disappears immediately and the application doesn't become unresponsive.

I retested the problem on two different PCs and can reproduce the bug. One one PC I get even an immediate crash on opening the DataPoint filter.

I don't have time right now for a bisect. However, this this there is no bug:

No problem with:

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.29298 (Git)
Build type: Release
Branch: master
Hash: 8573d9f8b258e59b5b357892ebed011aa2afbbba
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
```

And with this I get an immediate crash because of "illegal storage access":

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.29331 (Git)
Build type: Release
Branch: master
Hash: d13db0ea4c19a985a73a7bc30a24fe0598a21f6b
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2

```

wwmayer, 2022-06-28:
I have tested my Windows build (msvc) and there I can confirm the crash. With e07cfa09bb I have extended the unit tests that demonstrates the crash, too.

The crash is triggered by the call of `PropertyEnumeration::Paste(const Property &from)` because for some reason the `_enum` member has dangling pointers to the `Model` instances (see Enumeration.h).

The fact that the pointers become dangling is very weird because nowhere in the Enumeration class raw pointers are used but std::shared_ptr. And it's up to the std::shared_ptr class to increment an internal counter when assigning a std::shared_ptr to another instance of std::shared_ptr and this apparently doesn't work with msvc.

For testing purposes go to `Enumeration::operator=` and add this block
```
    int num = 0;
    for (const auto&it : other.enumArray)
        num = it.object.use_count();
```
before assigning the `enumArray = other.enumArray;` and add it again after the assignment. Now run the debugger and execute this Python snippet:
```
doc = App.newDocument()
enumeration_choices = ["one", "two"]
obj = doc.addObject("App::FeaturePython","Label_2")
obj.addProperty("App::PropertyEnumeration", "myEnumeration", "Enum", "mytest")

obj.myEnumeration = enumeration_choices
obj.myEnumeration = 0
doc.openTransaction("Modify enum")
obj.myEnumeration = 1
doc.commitTransaction()
doc.undo()
```
When running through the for-loop the first time then num is 1 for each entry which is correct. However, in the second for-loop it is still set to 1 for each entry and this is clearly wrong because it should be 2 instead.

This explains why the the pointers become dangling.

Now when testing it under Linux everything works as expected. So, it might be a bug in the implementation of std::shared_ptr of msvc.

wwmayer, 2022-06-29:
Renamed the title because after all it's not related to FEM at all.

donovaly, 2022-07-02:
Many thanks!
If there is a bug in MSVC I would like to report it. Astonishingly for MSVC Microsoft is listening to the community and fixed issues I too suffered from within weeks.
So if you have a small demo program, I can report the bug.

wwmayer, 2022-07-03:
[type_erasure.cpp.txt](https://github.com/FreeCAD/FreeCAD/files/9034172/type_erasure.cpp.txt)

commit 9a70b165 (2022-06-29): App: fixes #7102: Undo/Redo of PropertyEnumeration fails with MSVC