FreeCAD/FreeCAD #7112: [Bug] Addon manager does not work on windows with special characters in the username

wwmayer, 2022-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=606058#p606058

### Version

0.20 (Release)

### Full version info

```shell
.
```

### Subproject(s) affected?

Addon Manager

### Issue description

https://forum.freecadweb.org/viewtopic.php?p=606058#p606058

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2022-06-30:
@wwmayer is your commit above supposed to completely solve the issue?

wwmayer, 2022-07-02:
> is your commit above supposed to completely solve the issue?

It fixes the specific issue for Metadata but maybe there is somewhere else a problem, too.

chennes, 2022-07-06:
Is there still another issue? The rest of the Addon Manager is in pure Python, so I expect it to work now that the Metadata code has been fixed.

luzpaz, 2022-07-26:
Is this still an issue?

wwmayer, 2022-07-29:
> Is this still an issue?

Yes. I tested it yesterday and it still doesn't work in all cases. I think for local 8-bit it works now but not for real unicode path names. This can be easily tested without having to create a new account with a unicode string, just set the environment variable FREECAD_USER_HOME accordingly:
```
set FREECAD_USER_HOM=D:/Projects/ウィキペディアへようこそ/AppData/Roaming/FreeCAD
```
If you start FreeCAD and install any addon the files will be extracted to the correct directory but an XML exception is raised that something is wrong with the content of package.xml. If the file is moved to a directory  with 7-bit ASCII characters only the file can be loaded, so its content is valid.
The error messages is (see the "?"):
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
Base.XMLBaseException: Fatal error at file "", line 0, column 0
   Message: unable to open primary document entity 'D:/Projects/????????????/AppData/Roaming/FreeCAD/Mod/Curves/package.xml'
```

But what's even worse is when you restart FreeCAD. First it starts normally and shows the splash screen but because of the raised XML exception the GUI init script will be aborted and the calling instance will quit before starting the event loop.
So, there are two issues to fix:
1. Make sure to use the right parse() function of the xerces library
2. At least an XML exception caused by the metadata should be handled within the GUI init script to not abort the application

commit 9b165e6f (2022-07-29): App: fixes #7112: Addon manager does not work on windows with special characters in the username
commit 130da192 (2022-07-29): App: fixes #7112: Addon manager does not work on windows with special characters in the username
closed 2022-07-29 by wwmayer