FreeCAD/FreeCAD #7114: [Feature Request] Library of threaded PVC pipe adapters for folks in the US

bhs67, 2022-06-29:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=69504&sid=c094f003e5fc2bb727d55f351b30e878&start=10

### Idea description

I go to the hardware store and buy a PVC 1 1/2 inch male adapter.   The threaded diameter is slightly smaller on the front end, so it tightens as I screw it on a 1 1/2 inch female adapter.

All PVC pipes and threaded adapters are the same throughout the United States. Whatever it is, I buy it and everything fits.

It would be good if there were a standard library of threaded PVC pipe adapters for US folks.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-06-29:
This is more a [Fasteners workbench](https://github.com/shaise/FreeCAD_FastenersWB) thing. Please don't open a ticket without FC veterans in the community giving the green light.

closed 2022-06-29 by luzpaz

bhs67, 2022-06-29:
I did not know not that FC veterans approval is a requirement.