FreeCAD/FreeCAD #7125: [Bug] FreeCAD.Units.Quantity broken after changing Draft snapRange on-the-fyl

Roy-043, 2022-07-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=69886

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 8.1 (6.3)
Word size of FreeCAD: 64-bit
Version: 0.21.29283 (Git)
Build type: Release
Branch: master
Hash: 15e3e86c5a683f3b2e5c7b95d4b26cd719ab367a
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Draft

### Issue description

Changing the Draft snapRange on-the-fly seems to break FreeCAD.Units.Quantity.

Test scenario:

1. Start FreeCAD.
2. Switch to the Draft WB.
3. Change the gridSpacing to 1 mm (Edit → Preferences... → Draft → Grid and snapping → Grid → Grid spacing)
4. Open a new document.
5. Paste: `FreeCAD.Units.Quantity("1,00 mm")` in the Python console.
6. Result: `1.0 mm`. OK.
7. Make sure the grid is visible ([Draft_ToggleGrid](https://wiki.freecadweb.org/Draft_ToggleGrid))
8. Start the [Draft_Line](https://wiki.freecadweb.org/Draft_Line) command.
9. Pick the first point in the 3D view.
10. Click the `[` key to increase the snapRange (see [Draft_Snap#Preferences](https://wiki.freecadweb.org/Draft_Snap#Preferences)).
11. Move the cursor a little bit.
12. Result: the grid disappears.
13. Pick a second point to finish the command.
14. Paste: `FreeCAD.Units.Quantity("1,00 mm")` in the Python console.
15. Result: `2.2250738585072014e-308`.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2022-07-04:
I can't reproduce this... It still gives me `1.0 mm` after I do the above steps, and the grid doesn't disappear either..

bgma, 2022-07-05:
I can confirm the result "2.2250738585072014e-308"

```
OS: Windows 7 SP 1 (6.1)
Word size of FreeCAD: 64-bit
Version: 0.21.29393 (Git)
Build type: Release
Branch: master
Hash: 6820e0a9ec85203a6f342ca72a2ff8fd417beaf1
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: Swedish/Sweden (sv_SE)
Installed mods: 
  * fasteners 0.3.46
```

Roy-043, 2022-07-10:
Could this be Coin 4.0.0 related?
See https://github.com/FreeCAD/FreeCAD/issues/6767

Roy-043, 2022-07-11:
Hmm, apparently it is not Coin 4.0.0 related. I can reproduce Coin 4.0.1.
```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
  * BIM 2021.12.0
  * CubeMenu
  * Curves 0.4.4
  * fasteners 0.3.44
  * Help 1.0.3
  * Importer3D
  * InventorLoader 1.3.0
  * parts_library
  * ProDarkThemePreferencePack 1.0.0
  * TemplatePyMod
```

Roy-043, 2022-07-11:
FWIW the problem also exists in V0.18 and V0.19. Although in those versions the grid does not disappear as it does not use units for the gridSpacing.

```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
```

```
OS: Windows 8.1
Word size of OS: 64-bit
Word size of FreeCAD: 32-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: Dutch/Netherlands (nl_NL)
```

Roy-043, 2023-11-24:
I can reproduce this issue on Ubuntu as well. The only difference is that the grid disappears after clicking the 2nd point. There seems to be a relation with the shortcut key. If I use `F` instead of `[` the problem does not occur.

```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34836 (Git)
Build type: Release
Branch: main
Hash: 862eef57eb4aa042f87d9f19dc768a017b17dae6
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

Roy-043, 2025-12-29:
Still an issue:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.2.0dev.20251223 (Git shallow) AppImage
Build date: 2025/12/23 18:19:32
Build type: Release
Branch: (HEAD detached at 065fea4)
Hash: 065fea473f080691988bb11092759bf40f3c46fe
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```