FreeCAD/FreeCAD #7134: Qt6: Add Clazy-based tool to CMake to facilitate porting from Qt5 to Qt6

luzpaz, 2022-07-04:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=65089

### Idea description

There is a Clazy-based tool available to facilitate porting from Qt5 to Qt 6: [Porting C++ Applications to Qt 6 using Clazy checks](https://doc.qt.io/qt-6/porting-to-qt6-using-clazy.html). Specifically https://doc.qt.io/qt-6/porting-to-qt6-using-clazy.html#if-using-cmake 

Ref: #6992

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-07-04:
from my limited understanding of this it is something that you have to run locally, not something that needs to be added to the source code, so I'm not sure what you are asking for here or why make a separate issue from the qt6/pyside6 port issue?

luzpaz, 2022-07-04:
Maybe we can have a dedicated CI testing this for a possible qt6 branch?

adrianinsaval, 2022-07-04:
Could be, not sure if this is something that would be meaningful for every commit, this shouldn't be setup until there is some minimum support for qt6 though.

luzpaz, 2022-07-08:
Thanks for the feedback adrian.

@chennes what do you think? Maybe we can coordinate with @berndhahnebach to flag Qt6 builds to trigger compiling with this Clazy porting tool ?

chennes, 2022-07-08:
I don't understand the workflow you are envisioning here: can you elaborate?

adrianinsaval, 2022-07-08:
As chennes said in the forum there's nothing to do until we actually have a workable PR porting to qt6, it's not even possible to attempt building with qt6 right now and running the clazy checks on ci while nobody is working on the matter doesn't make sense either. This should be run locally by someone interested in it, the docs says the changes clazy does make it not compilable with qt5, only qt6, so after running it someone has to manually put those changes inside if else blocks.
What could be done is to prepare for the future and add qt6 stuff in the dockerfile for the ci

luzpaz, 2022-07-08:
Workflow: Some sort of way to distinguish Qt6 PRs (possibly using github labels?) so that when bernd's CI recognizes them as Qt6 related, will automagically tweak the build process (CMake) to accommodate

adrianinsaval, 2022-07-08:
Once we have working code IMO it would be easier to just have the CI include qt6 deps, then in qt6 PR you make qt6 the default and when merging the commit making qt6 default is skipped. This way we don't need any convoluted way of running the CI differently.