FreeCAD/FreeCAD #7142: Part Design Pad tool does not work (co-planar issue?)

FreeCAD-Bug-Importer, 2022-07-06:


adrianinsaval, 2022-07-06:
Also doesn't work on archlinux's 0.19.4:
```
OS: Linux 5.18.9-arch1-1 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24415 (Git)
Build type: Release
Branch: makepkg
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.10.5
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```
Of all my test it only worked in the 0.19 flatpak when it used occt 7.3 so it could be a regression upstream

grubertechnic, 2023-07-23:
@adrianinsaval : can you retest it? For me it works flawless.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33631 (Git)
Build type: Release
Branch: master
Hash: 2167fb6f7ec6ac5c2f6365d4c3035d2db6879b1e
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

I add the file direct here.

[LedKehysVaunuun_test.zip](https://github.com/FreeCAD/FreeCAD/files/12138512/LedKehysVaunuun_test.zip)

adrianinsaval, 2023-07-23:
I still experience this, might be fixed in occt 7.7.1, will try a version with 7.7.1 later today
```
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33593 (Git)
Build type: Release
Branch: master
Hash: 15371a04f38bbeb5a04edc5a91020ada1e30ef88
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Paraguay (es_PY)
```

grubertechnic, 2023-07-23:
Thanks. Note, that OCCT have created a 7.7.2 head, which means that in a certain time (normally some days) they release an update.

adrianinsaval, 2023-07-23:
Indeed, works without issue with occt 7.7.1:
```
OS: KDE Flatpak runtime
Word size of FreeCAD: 64-bit
Version: 0.21.0.33439 (Git)
Build type: Release
Branch: (HEAD detached at 976ecb3)
Hash: 976ecb3754700dfe6059699b77f896f9610891e3
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.7.1
Locale: English/United States (en_US)
```

adrianinsaval, 2023-07-23:
I'll close the issue since this is an upstream issue so not actionable by us, and it's fixed upstream so it will eventually make it's way to our releases once we get full 7.7.1 compatibility

closed 2023-07-23 by adrianinsaval

luzpaz, 2023-07-23:
Maybe we should add this as a 'Known Issue' that is solved upstream ? (especially for the distros that don't keep OCC as up to date as we do?)

adrianinsaval, 2023-07-23:
Where do we have such a list?

luzpaz, 2023-07-23:
https://forum.freecad.org/viewtopic.php?f=3&t=30573

adrianinsaval, 2023-07-23:
I think we use that for different general usage problems, mentioning specific opencascade coplanar-like problems seems like it will quickly fill the page with noise