FreeCAD/FreeCAD #7179: [Feature Request] PD: add more options in revolve/groove feature (two angles, up to face/fisrt/last)

69Piero, 2022-07-13:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

PD: add more options in revolve/groove feature (two angles, up to face/fisrt/last)

### Subproject(s) affected?

PartDesign

### Idea description

it's useful for the design adding some options in revolve/groove feature like in pad/pocket one. I mean:

- Two directions (two different angles from revolving axis)
- up to face/first/last

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-07-13:
Closing ticket because a forum thread wasn't opened (per guidelines in ticket) Once one is, and discussion leads to a green light, ticket can be reopened

AjinkyaDahale, 2022-07-13:
> Closing ticket because a forum thread wasn't opened (per guidelines in ticket) Once one is, and discussion leads to a green light, ticket can be reopened

https://forum.freecadweb.org/viewtopic.php?f=8&t=70185&sid=e8791a85acf23d80d2c532406ace0430

69Piero, 2022-07-13:
here is the thread

https://forum.freecadweb.org/viewtopic.php?p=609439#p609439

AjinkyaDahale, 2022-07-13:
@luzpaz you didn't reopen this.

PR #7193 (2023-11-20): PD: more options for revolution/groove
Attempts to fix #7179.

I am trying to copy the code from pad/pocket but they are somewhat different.