FreeCAD/FreeCAD #7188: [Bug][FEM] test_constraint_transform_torque fails on aarch64 due to minor inaccuracies

StefanBruens, 2022-07-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

-

### Version

0.20 (Release)

### Full version info

```shell
FreeCAD is run in an automatic build environment. No interactive access possible.
```

### Subproject(s) affected?

FEM

### Issue description

```
[  177s] FAIL: test_constraint_transform_torque (femtest.app.test_solver_calculix.TestSolverCalculix)
[  177s] ----------------------------------------------------------------------
[  177s] Traceback (most recent call last):
[  177s]   File "/usr/lib64/FreeCAD/Mod/Fem/femtest/app/test_solver_calculix.py", line 278, in test_constraint_transform_torque
[  177s]     self.input_file_writing_test(get_namefromdef("test_"))
[  177s]   File "/usr/lib64/FreeCAD/Mod/Fem/femtest/app/test_solver_calculix.py", line 399, in input_file_writing_test
[  177s]     "CalculiX write_inp_file for {0} test failed.\n{1}".format(base_name, ret)
[  177s] AssertionError: 'Comparing /usr/lib64/FreeCAD/Mod/Fem/femtest/data/calculix/constraint_transform_torque.inp to /tmp/FEM_unittests/solver_calculix_constraint_transform_torque_050998eb9bca/Mesh.inp failed!\n--- \n+++ \n@@ -11717 +11717 @@\n-2505,2,-9.5670268990152E-01\n+2505,2,-9.5670268990153E-01\n@@ -12539 +12539 @@\n-3327,2,-2.2462134450621E+00\n+3327,2,-2.2462134450620E+00\n' is not false : CalculiX write_inp_file for constraint_transform_torque test failed.
[  177s] Comparing /usr/lib64/FreeCAD/Mod/Fem/femtest/data/calculix/constraint_transform_torque.inp to /tmp/FEM_unittests/solver_calculix_constraint_transform_torque_050998eb9bca/Mesh.inp failed!
[  177s] --- 
[  177s] +++ 
[  177s] @@ -11717 +11717 @@
[  177s] -2505,2,-9.5670268990152E-01
[  177s] +2505,2,-9.5670268990153E-01
[  177s] @@ -12539 +12539 @@
[  177s] -3327,2,-2.2462134450621E+00
[  177s] +3327,2,-2.2462134450620E+00
```

Obviously, the two affected nodes have the same coordinates, differing only in a float LSB. Unfortunately, the code only compares the values verbatim, not taking  into account rounding/accurary of floats.

### Anything else?

Complete build log is available here:
https://build.opensuse.org/package/live_build_log/science/FreeCAD/openSUSE_Leap_15.3/aarch64

Test output:
https://build.opensuse.org/package/live_build_log/science/FreeCAD-test/openSUSE_Leap_15.3/aarch64

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-07-15:
@StefanBruens this is building v0.20 stable or HEAD ?

StefanBruens, 2022-07-15:
This is 0.20 stable/release, but at least the python and testing side of the code is unchanged in HEAD.
There may be changes in the writer code which would affect the result, but even then checking for *exact* match of floating point values is never a good idea. Minor (~LSB) differences between different architectures is a recurrent pattern.

Will also resurrect my HEAD builds to verify.

StefanBruens, 2022-07-18:
Just confirmed, same issue with HEAD of yesterday (2022-07-18, SHA 5048f14d).
https://build.opensuse.org/package/live_build_log/home:StefanBruens:branches:science:next/FreeCAD-test/openSUSE_Tumbleweed/aarch64

coldtobi, 2022-08-18:
I see those failures also in Debian, on the architectures
  arm64, i386, ppc64el and s390x
more infos (e.g build logs) on the Debian CI site: https://ci.debian.net/packages/f/freecad/
(This is on 0.20.0)

luzpaz, 2022-08-18:
@donovaly putting this on your radar, if that's OK?

donovaly, 2022-08-22:
@StefanBruens , is the issue still there in current master?

vkhodygo, 2023-02-23:
I just attempted to build [5f3e781](https://github.com/FreeCAD/FreeCAD/commit/5f3e781dbf280c5f8c97183414e2aeb93a57b0be) on my `x86_64` machine, no result so far.

mrtimdog, 2023-03-11:
Hi, I've been hitting this one on master (last attempt with d0c5b7c045758809a43708fefb9f7c6cf9fe650b).
I'm seeing it with -O3, but not -O2. I'm also using CFLAGS="-march=skylake-avx512 -mtune=skylake-avx512".

StefanBruens, 2023-03-12:
> Hi, I've been hitting this one on master (last attempt with [d0c5b7c](https://github.com/FreeCAD/FreeCAD/commit/d0c5b7c045758809a43708fefb9f7c6cf9fe650b)). I'm seeing it with -O3, but not -O2. I'm also using CFLAGS="-march=skylake-avx512 -mtune=skylake-avx512". EDIT: Hmm, this is odd, I'm going through combinations of gcc options trying to narrow down a trigger for this (it's going to take quite some time due to my long covid, so hopefully I'll get there!).

This is probably caused by FMA (Fused-Multiply-Add). This has higher precision as doing the multiplication and addition in separate instructions, as the multiplication result is rounded for non-FMA.

vkhodygo, 2023-03-13:
~~> FMA~~

~~Not necessarily, vectorization of your code sometimes leads to such effects. I'm not even sure `fma` is IEEE-compilant.~~

~~Non-associative properties of floating-point arithmetic can mess around with your results. We're lucky `-O3` by default doesn't change the order of operations unless you use `icc` though. These changes are pretty nasty and difficult to trace back.~~

Scratch that, this can easily be [anything](https://stackoverflow.com/questions/55974090/clang-gcc-only-generates-fma-with-ffast-math-why) (including vectorization), see your compiler specs. Also, do not compare string representations of floating-point numbers unless you're 100% sure this is what you need, introduce some tolerance to avoid that.

petterreinholdtsen, 2023-07-01:
Debian worked around this issue by disabling the test using the patch available from https://salsa.debian.org/science-team/freecad/-/blob/master/debian/patches/2030-fix-1017598.patch , see also https://bugs.debian.org/1017598 .

petterreinholdtsen, 2023-07-09:
I had a look at the code, and found the location where 'diff' is used to compare the real output with the expected output.

```
index 6e457e92a1..984d2efc55 100644
--- a/src/Mod/Fem/femtest/app/support_utils.py
+++ b/src/Mod/Fem/femtest/app/support_utils.py
@@ -208,6 +208,7 @@ def compare_inp_files(
         li.startswith("**   written ") or li.startswith("**   file ") or li.startswith("17671.0,1")
     )]
     lf2 = force_unix_line_ends(lf2)
+    # FIXME need way to compare floating point numbers using epsilon
     import difflib
     diff = difflib.unified_diff(lf1, lf2, n=0)
     result = ""
```

Not sure if the best approach is to rewrite the output from FemMesh::writeABAQUS in src/Mod/Fem/App/FemMesh.cpp or compare_inp_files() in src/Mod/Fem/femtest/app/support_utils.py?  What is your view on this?

petterreinholdtsen, 2023-07-09:
Would something like this b acceptable, reducing the number of digits after comma in the printout?

```
--- freecad-salsa.orig/src/Mod/Fem/App/FemMesh.cpp      2023-07-09 08:35:17.346123383 +0200
+++ freecad-salsa/src/Mod/Fem/App/FemMesh.cpp   2023-07-09 08:35:17.342123374 +0200
@@ -2137,6 +2137,13 @@
     // write nodes
     anABAQUS_Output << "** Nodes" << std::endl;
     anABAQUS_Output << "*Node, NSET=Nall" << std::endl;
+
+    // Print predictable and limited floating point number presision
+    // to get identical output on different CPUs, see
+    // https://github.com/FreeCAD/FreeCAD/issues/7188.
+    anABAQUS_Output << std::fixed;
+    anABAQUS_Output.precision(10);
+
     // This way we get sorted output.
     // See http://forum.freecadweb.org/viewtopic.php?f=18&t=12646&start=40#p103004
     for (VertexMap::iterator it = vertexMap.begin(); it != vertexMap.end(); ++it) {
```

luzpaz, 2023-10-13:
@FEA-eng @marioalexis84 can you weigh-in ?

vkhodygo, 2023-10-13:
@petterreinholdtsen 

I lean towards a better comparator. Employing tolerance is a valid scientific approach, whereas artificially truncating the output makes the results non-portable in that sense you won't be able to use them anywhere else outside the program.

petterreinholdtsen, 2023-10-13:
[Vladimir]
> artificially truncating the output makes the results non-portable in
> that sense you won't be able to use them anywhere else outside the
> program.

I do not understand this statement.  How can '1.2345678901' with 10
fraction digits after the dop be unusable outside FreeCAD while
'1.23456789012' with 11 fraction digits be usable?

-- 
Happy hacking
Petter Reinholdtsen

StefanBruens, 2023-10-13:
> [Vladimir]
> artificially truncating the output makes the results non-portable in that sense you won't be able to use them anywhere else outside the program.
> I do not understand this statement. How [...]

Because truncating does not solve the problem, it just makes it less probable to appear.

As long different machines may round slightly differently (due to accumulation of errors, and possibly different evaluation order), there will always be cases where one machine may round up, and the other round down, and rounding may affect an arbitrary number of digits.

I.e. one machine will have a result of e.g. 1.9999999999998, and the other one 2.00000000003, and truncation leaves you with different results. There are cases where truncation may suppress the rounding differences (e.g. with 2.00000001 and 2.00000003), but using a relative tolerance is definitely better.

PR #13292 (2024-04-08): FEM: Float comparison for FEM tests
This is an attempt to close #7188 and resolve issues in #12985

Still using `difflib.unified_diff` to check files, but this PR adds a more thorough check for lines that differ.

1. Lines are split into elements by using different strategies, as there are different file formats. Then these elements are converted to float
2. Then the lines are compared element-wise as floats with some absolute tolerance (`1e-10`)
3. If the lines are equal element-wise, they are removed from the lines that differ

I felt a strong urge to write some tests to better document the  [... cut]
closed 2024-04-08 by yorikvanhavre