FreeCAD/FreeCAD #7189: [Bug] Pad "Up to Last" can end too soon

AjinkyaDahale, 2022-07-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

TODO

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.29506 +1 (Git)
Build type: Release
Branch: pd-more-revol-options (No changes to pad)
Hash: 197797fc76cb77dff296d694844ce8e4dcb366fc
Python 3.10.5, Qt 5.15.5, Coin 4.0.1, Vtk , OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.4.4
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

If a looping face intersects a pad twice in a single direction, once as the last intersection, it only goes till the first instance instead of all the way.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

master-puller, 2022-07-15:
@AjinkyaDahale Can you provide the file?
From my experience, it's much faster and easier to fix when you can validate behaviors with a testcase

luzpaz, 2022-07-15:
I've attached the attached forum file to this ticket.

adrianinsaval, 2023-05-04:
I don't know how we could fix this, occt is doing what is being asked by the code: pad up to the intersection with the elliptic surface,

AjinkyaDahale, 2023-05-04:
> I don't know how we could fix this, occt is doing what is being asked by the code: pad up to the intersection with the elliptic surface,

The solution is to tell OCCT something else then. I haven't looked at this in a while, but depending on what options we have available this may become an upstream issue. If it holds us back this can be removed from the project and added to "known issues".

luzpaz, 2023-05-05:
So the way forward is to make a temporary workaround and also open an upstream OCC ticket ?

adrianinsaval, 2023-05-05:
> The solution is to tell OCCT something else then. I haven't looked at this in a while, but depending on what options we have available this may become an upstream issue. If it holds us back from releasing 0.21 this can be removed from the project and added to "known issues". Then if the solution within freecad is found, it can be backported.

Not an upstream issue IMO, they don't have an "up to last" option, only what would be the equivalent of up to face, for up to last we check for all face intersections in the pad direction and use the last intersected face for an up to face pad. The problem is that there is a previous intersection of the same face that freecad doesn't take into account.

https://github.com/FreeCAD/FreeCAD/blob/c5748fde7b5b975cab637ab99a016db93d882a85/src/Mod/PartDesign/App/FeaturePad.cpp#L160
https://github.com/FreeCAD/FreeCAD/blob/c5748fde7b5b975cab637ab99a016db93d882a85/src/Mod/PartDesign/App/FeatureExtrude.cpp#L203-L208
https://dev.opencascade.org/doc/refman/html/class_b_rep_feat___make_prism.html#ab52a53c45c3d1d62433a1c963f04ac93

What I'm thinking now is that for up to last we could instead make a pad up to the bounding box of the base feature then translate the profile to the end of that pad and make a pocket up to first in the opposite direction.

regarding 0.21, I'm inclined to mark this as out of scope.

> So the way forward is to make a temporary workaround and also open an upstream OCC ticket ?

I don't think it would be temporary and there's nothing to report to them, their API is behaving as it should, we tell it pad up to this face, it does that.
I don't know what's occt's policy for feature requests but I suspect if there isn't cash attached to it it's pointless.

FlachyJoe, 2023-05-06:
Possible solution:
 - check if face is cut twice (boolean intersection with sketch normal give two points)
 - if so divide the face per a plane passing throw its center and parallel to the sketch
 - use only the far/near part for up to last/first

adrianinsaval, 2023-05-07:
That only works for cases like this, bspline surfaces will be problematic plus it might very well be that the middle and parallel to sketch is not a good place to split, imagine if that ellipse was at 45deg instead

adrianinsaval, 2023-06-05:
Here's a file showcasing the algorithm that I believe we can follow for these cases:
[algorithm showcase.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11647611/algorithm.showcase.FCStd.zip) (remove zip extension)

if last intersected surface is intersected more than once:
• extrude bounding rectangle of profile past body bounding box in the selected direction
• boolean common between extrusion and last surface
• the operation should result in at least two faces, pick the one furthest from the profile
• extrude profile past body bounding box
• slice apart extrusion using the face picked earlier, discard all solids beyond this face and use the rest as addsubshape
• perform union between basefeature and addsubshape to obtain new shape

adrianinsaval, 2023-06-05:
Or another possibility that might by simpler is to translate the profile outside the body bounding box in the extrusion direction, then make an up to first extrusion in the opposite direction and subtract this from an extrusion of the original profile

FlachyJoe, 2023-06-08:
I've found another related bug, see this central object:
[image]

AjinkyaDahale, 2023-06-08:
> I've found another related bug, see this central object: 

@FlachyJoe could you elaborate what you tried to do here? Ideally add an example file and the FC version used.

adrianinsaval, 2023-06-08:
This curious case works as desired using up to last, but has the opposite problem when using up to face:
[image]
[triple-intersection-example.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11692959/triple-intersection-example.FCStd.zip) (remove zip extension)

IMO when choosing up to face the desired behavior is that it stops on first intersection with the face

maxwxyz, 2023-10-20:
IMO it should not end at the first intersection with a parallel extrusion of the profile sketch plane but align and join with the whole face curvature, as this is done in any other CAD application. Same for "to first" and "to face".
New Issue: https://github.com/FreeCAD/FreeCAD/issues/11111

PR #9743 (2023-08-21): PartDesign: Fix uptolast ends too soon
Fix #7189 in many cases but still fails if the sketch has a hole at it center of gravity.
[image]

Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md)  [... cut]