FreeCAD/FreeCAD #7214: [Feature Request] delay in pre-selection highlighting

Zolko-123, 2022-07-18:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=70139

### Subproject(s) affected?

Core

### Idea description

https://forum.freecadweb.org/viewtopic.php?f=10&t=70139#p611105

> In order to find the object underneath the current mouse position a SoRayPickAction must be performed on the scene graph and the shape nodes must deliver each point, line segment or triangle in order to find the intersection points. This is a very expensive operation and for objects with 100,000 or millions of triangles this can be very slow. For the pre-selection this is even worse because the operation is performed for every little mouse move event.

for very large assemblies, I have to disable pre-selection highlighting because the system becomes so slow as to be close to un-usable when simply rotating and moving the model in the 3D window. Unfortunately, the selection itself becomes much more difficult when the pre-selection is not available.

Would it be possible to add a slight delay so that the objects under the mouse are highlighted only after some time ? Something like 0.1s of still mouse might make a huge difference, or even a setting in the preferences.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-12-21:
where is the selection highlighting located in the source code ?

Zolko-123, 2022-12-23:
@luzpaz :

> where is the selection highlighting located in the source code ?

I'm not sure, but : `grep ighlight * | grep election` in ~/src/Gui lists the files  _SoFCSelectionAction.cpp_, _SoFCSelection.cpp_, _SoFCUnifiedSelection.cpp_, _SoFCSelectionContext.cpp_ as possible candidates. 

I'd bet on _SoFCSelectionAction.cpp_. In there, the function `beginTraversal()` is the one that probably should be delayed. The problem is, I don't see where this function is called.

luisDias89, 2024-10-27:
@maxwxyz @luzpaz In my opinion, this issue should be in milestone 1.0 because it blocks the software.

maxwxyz, 2024-10-27:
@luisDias89 cannot be, this is an issue which existed a long time now, in every stable version.