FreeCAD/FreeCAD #7237: [Feature Request] Slot Counterbore

xsuper9988, 2022-07-23:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=19&t=70239

### Idea description

For faster and more convenient drawing Slot Counterbore

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2022-07-24:
Please fill up "subprojects affected" (that would be part design) and add a representative image.

xsuper9988, 2022-07-24:
> 

Sorry this is my first feature request
I have modified it again
Let me know if there is anything missing
I will add it asap
Thank you very much

luzpaz, 2022-07-25:
Why is there mention of solidworks here?

xsuper9988, 2022-07-25:
> Why is there mention of solidworks here?
Sorry I thought to provide more references
sorry i removed