FreeCAD/FreeCAD #7245: [Bug] An external Link to a Part will not adopt the Std_DrawStyle Wireframe override

xsuper9988, 2022-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?p=612317#p612317

### Version

0.21 (Development)

### Full version info

```shell
FreeCAD_weekly-builds-29485-2022-07-21-conda-Windows-x86_64-py38

OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.21.29485 (Git)
Build type: Release
Branch: master
Hash: a236ca843fdd6674afb6d7ed1454fbd3b547f5ea
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: Chinese/Taiwan (zh_TW)
Installed mods: 
  * A2plus 0.4.56a
  * Assembly3 0.11.3
  * Assembly4 0.12.2
  * Curves 0.5.0
  * Design456 0.0.1
  * DynamicData 2.46.0
  * ExplodedAssembly
  * fasteners 0.3.46
  * lattice2 1.0.0
  * Manipulator 1.4.9
  * parts_library
  * pcb
  * sheetmetal 0.2.49
```

### Issue description

1.DrawStyle_WireFrame
If the source object uses the external file Parts Link object
It will cause the DrawStyle_WireFrame mode to fail

2.DrawStyle_Display mode sync update
No Display mode sync update when file switching

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MiqCG, 2024-06-11:
I have encountered this bug with the **integrated assembly WB**: it affects the "insert component" tool too.

A part from "Wireframe draw style", the bug is also present in "Points draw style". Don't know if another issue is needed for that.

This issue is labelled with the "Part WB" tag but the same behaviour is found in linked bodies in Part Design.

The attached video is about two separate files, the first with inserted or linked components (same behaviour) inside an assembly, and the second file with copied parts in the assembly. This is because at the first moment I thought it was only an assembly related problem.

``
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git) AppImage
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * fasteners 0.5.22
``

https://github.com/FreeCAD/FreeCAD/assets/18265355/7df98880-9e3e-4842-b4b6-2df011180bfe

[Ass_Wire_View.zip](https://github.com/user-attachments/files/15795798/Ass_Wire_View.zip)

MiqCG, 2024-12-24:
Yes, there is still problems with DrawStyle override, at least in assemblies with inserted components:
https://github.com/user-attachments/assets/d1565977-11cf-4f3b-a4d7-7a3bb8b3f4f9

```
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * fasteners 0.5.33
```

MiqCG, 2025-05-26:
Yes, the bug is still present in the dev version
I tested the same file and you can see this new video:

https://github.com/user-attachments/assets/1a7313c7-3df4-4abb-abf0-8446c40908ed

`
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41837 (Git) AppImage
Build date: 2025/05/23 13:10:01
Build type: Release
Branch: main
Hash: a969f31573c23d0aa956e87a1528d4cc393468ba
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Catalan/Spain (ca_ES)
Navigation Style/Orbit Style/Rotation Mode: Revit/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.8695/1.07292
`

luzpaz, 2025-05-27:
@PaddleStroke any idea on how to fix this ?