FreeCAD/FreeCAD #7248: [Feature Request] A tool for kinematic analysis on sketches

FEA-eng, 2022-07-24:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=68826

### Subproject(s) affected?

Sketcher

### Idea description

Previous FR on this topic was closed for being to wide: #6908 This one is much more specific and definitely achievable, as described below.
A small tool for simple kinematic analysis on properly constrained sketches could be added. It would just require "driving" constraints in the sketch and reading positions of vertices, then calculating velocity and acceleration as derivatives of displacement and creating plots. It can be achieved with a simple script but such a script has to be rewritten for each sketch.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2022-07-25:
If you mean to use the sketcher solver as simulation then I believe this is achievable with assemly 4's animation feature

FEA-eng, 2022-07-25:
> If you mean to use the sketcher solver as simulation then I believe this is achievable with assemly 4's animation feature

It's cool if it works with sketches as well. However, animation is just a nice visual addition to kinematic analysis. What I'm talking about is being able to do some calculations - generate plots of displacement, velocity and acceleration vs time.

maxwxyz, 2024-03-30:
@PaddleStroke do you think this is feasible in Sketcher for  2D kinematics? Regarding your work on Assembly and MDDyn maybe this is also possible. Or should this be a separate WB?