FreeCAD/FreeCAD #7257: [Feature Request] When using Draft to mirror a Part can keep the same color

xsuper9988, 2022-07-26:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=23&t=69889

### Subproject(s) affected?

Draft

### Idea description

When I use Part for easy assembly
It will contain several Step files or several Badys
But when I use Draft to mirror a Part
His overall color will turn white

[image]

I tested on LinkBranch version
This version keeps its color the same
Hope someone can add this feature to the standard version
Thank you very much

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-02-11:
This is a regression introduced in v0.19.

Roy-043, 2023-02-11:
Oops, I got confused. This is not a regression.

PR #8427 (2023-02-11): [Draft] fix Draft_Mirror diffuse color issue.
Fixes #7257.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---