FreeCAD/FreeCAD #7263: [Feature Request] [TD]Curved Leader Lines

WandererFan, 2022-07-27:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=70558

### Subproject(s) affected?

Techdraw

### Idea description

Currently leader lines only allow segmented lines.  Curved lines are useful in many situations.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2022-07-28:
This may come down to the recurring request for being able to sketch in TechDraw.

luzpaz, 2023-11-06:
Looks related to #5693

FEA-eng, 2023-11-06:
> Looks related to #5693

#5693 could help solve this FR but it could be also solved independently. This one is much more specific.

luzpaz, 2023-11-17:
@WandererFan what would be involved in add this functionality. Care to elaborate at your convenience ?

WandererFan, 2023-11-17:
in DrawLeaderLine, add a property to remember which segments of the leader are lines and which are curves (Beziers maybe?), and something to remember if a waypoint is an actual point or a control point.

in QGTracker, add the ability to draw curved segments and return segment type along with waypoints.

in QGILeaderLine, add the ability to repaint curves based on the new properties.