FreeCAD/FreeCAD #7271: [Bug] Collada import loses textures

dgatwood, 2022-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?style=1&p=595287

### Version

0.20 (Release)

### Full version info

```shell
Version: 0.20
Revision number: 29177 (Git)
Release date: 2022/06/13 23:43:45
macOS 12.4 (misidentified as "10.16")
Word size: 64-bit
Branch: (HEAD detached at 0.20)
Hash: 68e3376
```

### Subproject(s) affected?

File formats

### Issue description

When importing any models downloaded from 3dwarehouse.sketchup.com, e.g.

https://3dwarehouse.sketchup.com/model/b008ecc8d46e346343da6f498cb41285/Industrial-dishwasher?hl=en

FreeCAD loses all of the textures.  The resulting exports are all grey.  This was reported by someone else on the forums a couple of months ago, with no real resolution as far as I can tell, other than to try to work around it by spending hundreds of dollars for SketchUp subscription and re-exporting it in a usable format.

Worse, FreeCAD is the only tool that even gets close to handling these files.  DAZ Studio loses several surfaces, Blender loses slightly fewer, Collada2GLTF loses one handle.  Only FreeCAD gets the structure completely right.  But without being able to translate the materials data, this huge treasure trove of great 3D models is effectively inaccessible to the general unless you're willing to spend hours hand-fixing all the materials on a per-component basis or hundreds of dollars to subscribe to SketchUp.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2022-07-29:
This statement is not exactly true, as [Blender](https://blender.org) has excellent support for Collada too. Furthermore, FreeCAD has ATM little to no support for textures, at least internally (some addons provided partial support). So in any case, there would be little you can do with textures in FreeCAD.
If you need to work with textures, I would suggest to use Blender instead, which has proper, extended and state-of-the-art tool for working with textures. (Note that there is good import/export between Blender and FreeCAD). Otherwise, unfortunately in FreeCAD itself, there is nothing we can really do at this point, as usable textures support is not something that will likely be integrated anytime soon.
One thing could be interesting, is to see if the materials defined in the collada file also define a color, apart from a texture. Then we could do something...
I will close this issue now because I don't think there is anything we can do to solve it, but feel free to reopen if needed.

closed 2022-07-29 by yorikvanhavre

dgatwood, 2022-07-29:
FreeCAD actually has enough support for materials to make it work well enough for my purposes.  That's actually how I was able to do a previous conversion of a different model.  But it was a painful process, because A. on one of the models, it had a whole-object surface that had to be omitted from the export, or else it "wins" over everything else, and B. you have to manually reassign each of the different surfaces to have a different material in FreeCAD before you export, because FreeCAD lumped them all into the same material during import.

The current released version of Blender had major problems with every .dae model I've tried so far (which is admittedly just a single-digit number of models).  Maybe Blender is still using an old build of pycollada.  No idea.  But it definitely fails to import these files in the latest released version (missing surfaces).

The only thing that handled this particular file completely correctly was MeshLab, which is a tool that I first learned about in the thread that led to this bug.

The limited support for materials in FreeCAD isn't a big deal, but I would expect the import to at least map different materials on input onto different FreeCAD materials so that you don't completely lose that structure when you import and re-export.