FreeCAD/FreeCAD #7276: [Bug] python time.sleep() freezes entire ui for duration of sleep when executing macro?

rydb, 2022-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?style=1&p=245811

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
[/code]
```

### Issue description

Using time.sleep() in python freezes FreeCAD for the duration of the sleep function.

**sample code:**

`print("Hello world!")
time.sleep(2)
print("Hello world!")`

**expected behavior:**
hello world runs once, pauses for 2 seconds, then runs again

**actual behavior**
FreeCAD freezes for 2 seconds, then both hello worlds are printed.

The linked thread does have a solution, but the solutions are lots of lines of code to do what normally takes one.

Is there a reason time.sleep() behaves the way it does?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Protowalker, 2022-08-01:
It's happening because the UI and the python runtime are on the same thread. A potential solution could be to just export a `FreeCAD.sleep()` that you can use instead of `time.sleep()`, which updates the UI and such

luzpaz, 2022-10-17:
See forum for solution: https://forum.freecadweb.org/viewtopic.php?style=1&f=22&t=29865

closed 2022-10-17 by luzpaz