FreeCAD/FreeCAD #7285: [Bug] FreeCAD crashes if cell value `'` is set in Spreadsheet

marzn, 2022-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: German/Germany (de_DE)
[/code]

and 

[code]
OS: Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: makepkg
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.10.5, Qt 5.15.5, Coin 4.0.1, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Issue description

1. Start FreeCAD
2. Switch to workbench Spreadsheet
3. Create a spreadsheet
4. Enter `'` in a cell and press enter
5. FreeCAD crashes

### Anything else?

##### Linux terminal

```
FreeCAD 0.20, Libs: 0.20R29177 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
[... 68 lines cut ...]
Log: Id: 3328: 00007FF8EB261A94 (Qt5Core): (filename not available): QCoreApplication::exec
Log: Id: 3328: ERROR: SymGetLineFromAddr64, GetLastError: 487 (Address: 00007FF8EB6959A8)
Log: Id: 3328: 00007FF8EB6959A8 (FreeCADGui): (filename not available): Gui::Application::runApplication
Log: Id: 3328: ERROR: SymGetSymFromAddr64, GetLastError: 487 (Address: 00007FF770183EEE)
Log: Id: 3328: ERROR: SymGetLineFromAddr64, GetLastError: 487 (Address: 00007FF770183EEE)
Log: Id: 3328: 00007FF770183EEE (FreeCAD): (filename not available): (function-name not available)
Log: Id: 3328: ERROR: SymGetSymFromAddr64, GetLastError: 487 (Address: 00007FF7701851F7)
Log: Id: 3328: ERROR: SymGetLineFromAddr64, GetLastError: 487 (Address: 00007FF7701851F7)
Log: Id: 3328: 00007FF7701851F7 (FreeCAD): (filename not available): (function-name not available)
Log: Id: 3328: ERROR: SymGetSymFromAddr64, GetLastError: 487 (Address: 00007FF7701847B6)
Log: Id: 3328: ERROR: SymGetLineFromAddr64, GetLastError: 487 (Address: 00007FF7701847B6)
Log: Id: 3328: 00007FF7701847B6 (FreeCAD): (filename not available): (function-name not available)
Log: Id: 3328: ERROR: SymGetLineFromAddr64, GetLastError: 487 (Address: 00007FF919D07034)
Log: Id: 3328: 00007FF919D07034 (KERNEL32): (filename not available): BaseThreadInitThunk
Log: Id: 3328: ERROR: SymGetLineFromAddr64, GetLastError: 487 (Address: 00007FF91B9A2651)
Log: Id: 3328: 00007FF91B9A2651 (ntdll): (filename not available): RtlUserThreadStart
Log: *** Unhandled Exception!
Log:    ExpCode: 0xC0000005
Log:    ExpFlags: 0
Log:    ExpAddress: 0x0E1F3698
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

markaidev, 2022-07-31:
Followed instructions and could not reproduce the crash using Ubuntu system.

luzpaz, 2022-07-31:
No forum discussion was opened for this ticket. Please do so

marzn, 2022-07-31:
Last time I tried to post on FreeCADs forum I got the following response:

> Your post is fine, so I should approve it. However, your username is not fine. Can you please chose something else as name that can be pronounced and s maybe shorter too?

I gave up after that response. May someone else can open a thread in the forum, can reproduce the issue or just close this bug report.

Just tested it on a third system with Ubuntu / Snap:
```
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) Snap 161
Build type: Release
Branch: tag: 0.20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: C/Default (C)
```

luzpaz, 2022-07-31:
marzn, sorry to hear that. Can you test development version in that case? Perhaps consider using [Snap edge](https://wiki.freecad.org/Snap) ?

marzn, 2022-07-31:
Snap edge crashes like the releases tested before.

```
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.29856 (Git) Snap 177
Build type: Release
Branch: master
Hash: a6fbd74caeed6c9f55041ee82ec227c12bc11900
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: C/Default (C)
```

luzpaz, 2022-07-31:
Thanks for testing. Opened thread: https://forum.freecadweb.org/viewtopic.php?t=70681

closed 2022-08-01 by wwmayer

marzn, 2022-08-01:
Thanks @wwmayer for your bug fix, it ignores the input `'` in a cell, which prevents FreeCAD from a crash.

As a user I would expect the input of `'` behaves like `=<<>>`, which is shown as `'` (what makes me assume, I should be able to enter it manually as well). But maybe this is a separate issue.

commit 2f197d65 (2022-08-01): Spreadsheet: fixes #7285: [Bug] FreeCAD crashes if cell value ' is set in Spreadsheet
commit 3a424136 (2022-08-01): Spreadsheet: fixes #7285: [Bug] FreeCAD crashes if cell value ' is set in Spreadsheet