FreeCAD/FreeCAD #7303: [Bug] opengl32.dll not named correctly

theleetbeagle, 2022-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=4&t=36087&start=20

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
[/code]
```

### Issue description

Well known issue that older computers, often with integrated graphics, will crash shortly after load or when creating new or opening file. 

Common suggestion to select Use OpenGL in preferences does not work. Users are often on the lastest installed drivers, which are several years old and discontinued. Submitter, for instance, is using a Dell E6510 laptop with latest drivers being from 2011. 

Using Build-0-Matic's suggestion on page 3 of this thread yields immediate results. 

https://forum.freecadweb.org/viewtopic.php?f=4&t=36087&start=20

Also, simply renaming opengl32sw.dll to opengl32.dll in the \bin\ directory fixes the issue.

Suggest altering FreeCAD installer to correctly rename opengl32.dll.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2022-08-04:
This is not a bug, if the file was renamed as proposed that would mean all Windows users would have to use Software Accelerated graphics rather than the (sometimes very expensive) hardware acceleration.The renaming of the file in it's present state is for those relatively few machines that their hardware/drivers aren't OpenGL compatible.

theleetbeagle, 2022-08-04:
> This is not a bug, if the file was renamed as proposed that would mean all Windows users would have to use Software Accelerated graphics rather than the (sometimes very expensive) hardware [...]

When a user is having an issue where the app is crashing, checking the OpenGL box under Preferences doesn't fix the issue. The user should not have to dive into directories to manually rename files, even if the issue were documented, which it isn't.

Seems like a bug to me.

adrianinsaval, 2022-08-17:
It's not a bug, the dll is purposefully named that because it is detrimental for the vast majority of people that have working graphics, this is provided as a workaround for the few with broken graphics and cannot be done in an automated way.

theleetbeagle, 2022-08-17:
Yeah, you keep saying that, except there's no documentation that a
user has to go into a damn folder and rename files to get their app to
work.

theleetbeagle, 2022-08-17:
Also, don't try to be a jerk w the working graphics comment. I have a Dell
latitude notebook. Hardly uncommon.

Your app is incompatible w it's drivers. I seriously doubt that's Intel's
fault.

luzpaz, 2022-08-23:
@wwmayer please weigh-in when you have a moment. TIA!

wwmayer, 2022-08-24:
It's definitely not a bug in FreeCAD because the crash happens in the graphic driver. And I agree that there is no point to make FreeCAD load the software-based opengl32.dll as the rendering may become very slow for all those systems where the system OpenGL version works flawlessly.

> Your app is incompatible w it's drivers.

How can it be as FreeCAD doesn't have its own drivers? It (indirectly) uses what's available on the system.

theleetbeagle, 2022-08-24:
So a toggle switch that doesn't actually do anything, and requiring
users to dive into directories to rename files (with no documentation
whatsoever saying you have to do so) is not a bug? Got it.

adrianinsaval, 2022-08-24:
> Also, don't try to be a jerk 

Please follow your own advice...

@wwmayer is it actually necessary to rename the DLL? Isn't it loaded when you select "use software OpenGL" in settings?

theleetbeagle, 2022-08-24:
@wwmayer is it actually necessary to rename the DLL?

Yes. Pay attention.

Isn't it loaded when you select "use software OpenGL" in settings?

No. Pay attention.

luzpaz, 2022-12-29:
What's the way forward here then ?

adrianinsaval, 2022-12-29:
Close the issue IMO, FreeCAD has no control over graphics driver

luzpaz, 2022-12-29:
@adrianinsaval can FC spit out a hint pointing to this ticket for folks encountering this issue?

adrianinsaval, 2022-12-29:
You mean an error message? Maybe

theleetbeagle, 2022-12-29:
> Close the issue IMO, FreeCAD has no control over graphics driver

FC has full control in giving the user notification / info that this issue exists.

adrianinsaval, 2022-12-29:
Ok let's reframe the issue then, at least the title should be changed. PRs welcomed, how can freecad detect that the graphics driver won't work? Should we suggest using this dll or enabling software opengl in settings?

adrianinsaval, 2024-01-26:
Stale issue and IMO out of scope for what we do, we don't have resources to do extensive hardware tests to know what works and what doesn't to be able to make actually useful suggestions to the users, and it doens't really make much sense to spend that much effort for discontinued hardware anyways.

closed 2024-01-26 by adrianinsaval