FreeCAD/FreeCAD #7304: [Bug] OpenSCAD: Imported Object is rotated 90° around X-Axis

gneiss15, 2022-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=10&t=70761

### Version

0.21 (Development)

### Full version info

```shell
FreeCAD_weekly-builds-29485-2022-07-24-conda-Linux-x86_64-py310
```

### Subproject(s) affected?

OpenSCAD

### Issue description

As written in the title, the imported Object is rotated.
(There are other import-bugs, but I was requested to file a separate report for each)

I had attached two scad-files.
The one that is rotated is the file "CR2032.scad".
The other one is imported in correct oriantation.

[Scad-Files.zip](https://github.com/FreeCAD/FreeCAD/files/9253596/Scad-Files.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2022-08-03:
Please explicitly add your full [About](https://wiki.freecad.org/About) to bug tickets. thank you!

luzpaz, 2022-12-01:
screenshot of bug would be nice.

chennes, 2022-12-02:
I believe this was fixed by 3f118b52bc62db6a5ca85daafaa2c021411e7177, but if not please re-open.

closed 2022-12-02 by chennes